Staubgeborener/Klipper-Backup #1: I can't get this script to work for me

simpat1zq, 2023-05-17:
I run the script, and this is what I get. I tried running the command it suggested, but that didn't work for me. Any ideas what I need to do?

pi@elf:~ $ ./klipper-backup/script.sh                                           Reinitialized existing Git repository in /home/pi/klipper-backup/.git/
fatal: pathspec '/home/pi/klipper-backup/.env' did not match any files
On branch master
nothing to commit, working tree clean
fatal: The current branch master has no upstream branch.
To push the current branch and set the remote as upstream, use

    git push --set-upstream https://MyGithubToken@github.com/simpat1zq/elf.git master

pi@elf:~ $ git push --set-upstream https://MyGithubToken@github.com/simpat1zq/elf.git master
fatal: not a git repository (or any of the parent directories): .git
pi@elf:~ $

Staubgeborener, 2023-05-21:
Sorry for the late reply, I am currently moving my apartment.

Is there a `.env` file in your `klipper-backup` folder and if so, did you edit the content inside so it match with your repository name and your GitHub token?

Please do a `ls -la /home/pi/klipper-backup` and post the result here.

simpat1zq, 2023-05-21:
Yes. The token is in the .env file. 

pi@coreception:~/klipper-backup $ ls -la
total 40
drwxr-xr-x  4 pi pi 4096 May 21 11:06 .
drwxr-xr-x 32 pi pi 4096 May 20 06:33 ..
-rw-r--r--  1 pi pi  866 May 17 12:39 .env
-rw-r--r--  1 pi pi  843 Apr  4 08:55 .env.example
drwxr-xr-x  7 pi pi 4096 May 21 11:06 .git
-rw-r--r--  1 pi pi   17 Apr  4 08:55 .gitignore
-rwxr-xr-x  1 pi pi 1902 Apr  4 08:55 install.sh
drwxr-xr-x  2 pi pi 4096 May 21 11:06 klipper
-rwxr-xr-x  1 pi pi  902 Apr  4 08:55 script.sh
-rw-r--r--  1 pi pi    5 Apr  4 08:55 version

pi@coreception:~/klipper-backup $ more .env
github_token=token goes here
github_username=simpat1zq
github_repository=coreception

path_printercfg=/home/pi/klipper_config/printer.cfg
path_mainsailcfg=/home/pi/klipper_config/mainsail.cfg
path_moonrakercfg=/home/pi/klipper_config/moonraker.conf
path_timelapsecfg=/home/pi/klipper_config/timelapse.cfg
path_telegramcfg=/home/pi/klipper_config/telegram.conf
path_crowsnestcfg=/home/pi/klipper_config/crowsnest.conf
path_adxlcfg=/home/pi/klipper_config/adxl.cfg
path_pacalcfg=/home/pi/klipper_config/pa_cal.cfg
path_getprobelimitcfg=/home/pi/klipper_config/GET_PROBE_LIMITS
.cfg
path_savedvariablescfg=/home/pi/klipper_config/saved_variables
.cfg
path_pa_flow_calibratecfg=/home/pi/klipper_config/pa_flow_cali
brate.cfg
path_soundscfg=/home/pi/klipper_config/sounds.cfg
path_macroscfg=/home/pi/klipper_config/macros.cfg

backup_folder=klipper
pi@coreception:~/klipper-backup $

Staubgeborener, 2023-05-28:
Try the following, step by step:

1. Create a repository on GitHub for your backup files (i think you already [did that](https://github.com/simpat1zq/elf))
2. Clone your repository: `cd ~ && git clone https://github.com/simpat1zq/elf`
3. Go into your repository: `cd ./elf`
4. Copy all files from this script in your own repository: `cp * ~/klipper-backup/ .`
5. Make the script executable: `chmod +x ./script.sh`
6. Run the script: `./script.sh`

If you get an error, please replace the line `git rm -rf --cached "$parent_path"/.env` with 
```
git filter-branch --force --index-filter \
  'git rm -r --cached --ignore-unmatch "$parent_path"/.env' \
  --prune-empty --tag-name-filter cat -- --all
```
in `script.sh`. I already did that with the commit [6dc586a](https://github.com/Staubgeborener/klipper-backup/blob/main/script.sh#LL28C1-L31C41). Try to re-run the script `./script.sh`.

Btw, i recognize, that the klipper paths have changed. So instead of `/home/pi/klipper_config/[...]` it should be `/home/pi/printer_data/config/[...]`. But check this for yourself before editing.

Staubgeborener, 2023-06-01:
I decided to create a [wiki](https://github.com/Staubgeborener/klipper-backup/wiki/) that provides detailed step-by-step instructions. 

In this concrete context, [this](https://github.com/Staubgeborener/klipper-backup/wiki/Installation%3A-Initialize-GitHub-repository) would be the interesting part for you.

closed 2023-07-16 by github-actions[bot]