Staubgeborener/Klipper-Backup #3: Entire printer_data folder support?

MallocArray, 2023-06-13:
Is it possible to backup the entire printer_data folder to a git repo, or must we define each individual file?

I was going to backup the entire folder and use .gitignore to exclude certain folders like logs that I don't need in a repo, but as I add more files, I don't want to have to remember to add them.

I attempted to add a folder to the configuration and it said I didn't specific `-R` but if I added that, the script blew up.

Staubgeborener, 2023-06-13:
`cp -r` is the way. I just tested it successfully. What exactly did you mean with "the script blew up"? Basically change `cd` to `cd -r` in [line 23](https://github.com/Staubgeborener/klipper-backup/blob/main/script.sh#L23).

Your `.env` has to look like this:
```shell
github_token=ghp_xxxxxxxxxxxxxxxx
github_username=USERNAME
github_repository=REPOSITORY

path_klipperdata=/home/pi/printer_data/

backup_folder=klipper
```
By the way, your approach with the `.gitignore` is also correct.

Staubgeborener, 2023-06-15:
`cp -r`

MallocArray, 2023-06-30:
I finally got back to this.  When I said the script blew up, this is what I saw
```
pi@mainsailos:~/klipper-backup $ ~/klipper-backup/script.sh
cp: -r not specified; omitting directory '/home/pi/printer_data/'
Reinitialized existing Git repository in /home/pi/klipper-backup/.git/
WARNING: git-filter-branch has a glut of gotchas generating mangled history
         rewrites.  Hit Ctrl-C before proceeding to abort, then use an
         alternative filtering tool such as 'git filter-repo'
         (https://github.com/newren/git-filter-repo/) instead.  See the
         filter-branch manual page for more details; to squelch this warning,
         set FILTER_BRANCH_SQUELCH_WARNING=1.
Proceeding with filter-branch...

Cannot rewrite branches: You have unstaged changes.
[main 3ce4f01] New backup from 29-06-23
```

After making the change on line 23 from `cp` to `cp -r` it now functions as I was hoping.  I'm in the middle of a print now, but since I edited the file, won't the moonraker updater page show a dirty repo since what I have doesn't match yours?

Staubgeborener, 2023-06-30:
> After making the change on line 23 from `cp` to `cp -r` it now functions as I was hoping. 

Nice!

> I'm in the middle of a print now, but since I edited the file, won't the moonraker updater page show a dirty repo since what I have doesn't match yours?

Sadly, that's correct. I need to fix that in order to give the end user the decision. An explanation of this problem can be found [here](https://github.com/Staubgeborener/klipper-backup/issues/4#issuecomment-1590680600):
> Actually the own backup repository should be in his own folder, which is independent from the klipper-backup folder - but while updating, klipper-backup must know exactly what this "own backup" folder is called and/or what the path is. I have to see which is the most elegant way.

Basically, it's the same type of issue which leads to this dirty repository message. If we just put the whole script in its own independent folder, the problem would be solved. But then I would have to see how to run updates - the script would have to know where this "independent folder" is (e.g. write the path to a dotfile which is somewhere on the file system? Something like this).

Staubgeborener, 2023-07-04:
About the "backup whole directory" part: I was overthinking this. Actually, it doesn't matter if we use `cp` or `cp -r` for this purpose - sure, `cp` only works for files (not directories) BUT `cp -r` works for directories **AND** files. So I [add](https://github.com/Staubgeborener/klipper-backup/commit/92646295dbab98e71521d9492b56b93817541541) the `-r` parameter as default in the code.

Staubgeborener, 2023-07-29:
Since the main problem is done (recursive copying as a feature) I'll close this. The dirty flag problem is a separate one.

closed 2023-07-29 by Staubgeborener

Staubgeborener, 2023-08-01:
> > 
> 
> 
> 
> This is not correct for some reason on any of my printers.... 
> 
> `cp: cannot stat '/home/pi/printer_data/': No such file or directory`

Because `/home/pi/printer_data/` doesn't exist on your device. You have to adjust the path inside the `.env` file: https://github.com/Staubgeborener/klipper-backup

lapluviosilla, 2023-10-03:
@Staubgeborener 
So, I tried this out, but if you try to use a subfolder then it recursively creates the files.
I put this in the .env
path_macrosdata=/home/pmex/printer_data/config/macros/
and 
the second time I ran the script it recursively create the folder so I ended up with
backup_folder/klipper/macros/macros/.

I fixed the issue by removing /$file on the copy command, the path to the backup folder is enough.

Staubgeborener, 2023-10-03:
@lapluviosilla Your right, this seems to fix this bug. I released a [new version 0.3.6](https://github.com/Staubgeborener/klipper-backup/releases/tag/0.3.6) which fixes the error.