Staubgeborener/Klipper-Backup #4: GitHub access token found in commit

jaakko000, 2023-06-13:
When i insert github_token into .env and push by "bash script.sh" inside ~/klipper-backup in SSH, github instantly revokes my access token. Is the repo supposed to be private or something? Doesnt seem to matter if its a fine grain or normal token.

Staubgeborener, 2023-06-13:
By "revoke" do you actually really mean that GitHub discards the token? So we are actually talking about the token being marked as "revoke" in your GitHub settings and not that you get simply am error message because the token is invalid (if so, please post the error message here)?

That's odd. The repo can be private or public. Sounds like [this behavior](https://docs.github.com/en/authentication/keeping-your-account-and-data-secure/token-expiration-and-revocation#token-revoked-when-pushed-to-a-public-repository-or-public-gist) tbh.

But that would mean that you either don't have a `.gitignore` file in the `klipper-backup` folder, or you have edited the `.gitignore`. This [prevents ](https://github.com/Staubgeborener/klipper-backup/blob/main/.gitignore#L1) the token from being pushed to GitHub respectively that the token will be revoked.

jaakko000, 2023-06-13:
Yes, discards the token. Goes away from the list of available tokens. Goes away if repo is public. That behavior would match imo. I have a .gitignore (only found with "ls -la"). Havent edited it and looks the same as yours. Seemingly works fine when i made the repo private. Weird. So maybe it's solved? Works almost perfectly now that I got past that hurdle. 

Just the moonraker updater klipper-backup is saying invalid and asking for soft or hard reset? Any ideas? Followed the guide

Staubgeborener, 2023-06-13:
Yeah, a private repository will work as GitHub security guidelines don't affect token from being revoked after pushing them. This only happens in public repository because of obviously reasons. But again, this is weird. Can you check your private repository on GitHub? Is the token somewhere in this repository in **any** file? My first guess was, that you put the token also in the .env.example (which will be pushed). It has to be somewhere. 

About the soft/hard reset issue: i think I now why this happens and I have a bad feeling about this. I will look into detail in a week. The macro should work anyway, I think you can simply ignore it for the time.

jaakko000, 2023-06-13:
> Can you check your private repository on GitHub? Is the token somewhere in this repository in **any** file? My first guess was, that you put the token also in the .env.example (which will be pushed). It has to be somewhere.

It is in .env. Nowhere else. I didn't get .env by following the guide and only got .env.example. Then I renamed that to .env and then put all that stuff in normally. Maybe it didnt change/register properly or something? How would I go about fixing that?

> About the soft/hard reset issue: 

Your explanation makes sense. Hope you can fix it. It works for the time being so i'll just run it manually. Commit history seemingly correct too as far as I can tell. Here is what running the script looks like. Your explanation here likely has to do with the "glut of gotchas" so I figured I'd post this. (newest at the top)

```
22.51
Command {update_git_script} finished
22.51
To https://github.com/jaakko000/Klipper_backup.git
ee8305d..199743d main -> main
22.51
[main 199743d] New backup from 13-06-23
23 files changed, 1772 insertions(+)
create mode 100644 klipper/config/config/.moonraker.conf.bkp
create mode 100644 klipper/config/config/crowsnest.conf
create mode 100644 klipper/config/config/fan_tach_monitor.cfg
create mode 100644 klipper/config/config/macros/Adaptive_Mesh.cfg
create mode 100644 klipper/config/config/macros/CG28.cfg
create mode 100644 klipper/config/config/macros/auto_z.cfg
create mode 100644 klipper/config/config/macros/auto_z_guide.cfg
create mode 100644 klipper/config/config/macros/klack.cfg
create mode 100644 klipper/config/config/macros/macros.cfg
create mode 100644 klipper/config/config/macros/nozzle_prime.cfg
create mode 100644 klipper/config/config/macros/parking.cfg
create mode 100644 klipper/config/config/macros/pauseresumecancel.cfg
create mode 100644 klipper/config/config/macros/power_plug.cfg
create mode 100644 klipper/config/config/macros/start_print.cfg
create mode 100644 klipper/config/config/macros/test_speed.cfg
create mode 100644 klipper/config/config/macros/update_git.cfg
create mode 120000 klipper/config/config/mainsail.cfg
create mode 100644 klipper/config/config/moonraker.conf
create mode 100644 klipper/config/config/printer-20230607_211240.cfg
create mode 100644 klipper/config/config/printer.cfg
create mode 100644 klipper/config/config/shell_command.cfg
create mode 100644 klipper/config/config/sonar.conf
create mode 120000 klipper/config/config/timelapse.cfg
22.51
index filter failed: git rm -r --cached --ignore-unmatch "$parent_path"/.env
22.51
Rewrite 78ee9915c9e22434f2686757456d8a072ae5c31d (1/20) (0 seconds passed, remaining 0 predicted) fatal: /.env: '/.env' is outside repository at '/home/pi/klipper-backup/.git-rewrite/t'
22.51
Proceeding with filter-branch...
22.51
WARNING: git-filter-branch has a glut of gotchas generating mangled history
rewrites. Hit Ctrl-C before proceeding to abort, then use an
alternative filtering tool such as 'git filter-repo'
(https://github.com/newren/git-filter-repo/) instead. See the
filter-branch manual page for more details; to squelch this warning,
set FILTER_BRANCH_SQUELCH_WARNING=1.
22.51
Reinitialized existing Git repository in /home/pi/klipper-backup/.git/
22.51
Running Command {update_git_script}...:
```

closed 2023-07-30 by github-actions[bot]