Staubgeborener/Klipper-Backup #9: Feature request: backup only if files changed

mrodyhin, 2023-07-25:
It would be nice to check the local repo state, and do the backup only if there is actual changes to backup. That will reduce number of commits and clear the repo history.

Staubgeborener, 2023-07-25:
Since the script is based on `git`, changes are only committed and pushed when changes are available. If the files (selected by the user himself) have not been changed, then these will also not been pushed and therefore not included in the commit history. Maybe you can explain what you mean in case I didn't get your point.

Taking a quick look into [my own backup commit history](https://github.com/Staubgeborener/3dprint/commits/main): If I run the script shortly after I pushed latest changes, leads to this message

```
On branch master
Your branch is up to date with 'origin/main'.

nothing to commit, working tree clean

Everything up-to-date
```

Which is the correct behavior.

mrodyhin, 2023-07-27:
it starts working after full reinstall, so closing this issue.

closed 2023-07-27 by mrodyhin