Staubgeborener/Klipper-Backup #10: Duplicate folders

RouvenWulff, 2023-07-31:
With the first commit, the files and folders are uploaded as expected:

```
Klipper (folder)
--configs (folder)
----klicky.cfg
----macros.cfg
----display.cfg
--KAMP (folder)
----adaptive_meshing.cfg
----adaptive_purging.cfg
printer.cfg
mainsail.cfg
moonraker.conf
sonar.cfg
crownsnest.conf
KAMP_Settings.cfg
```

After a reboot, the folders are duplicated and moved below the original folder. All files are duplicated, too

```
Klipper (folder)
--configs (folder)
----configs (folder)
------klicky.cfg
------macros.cfg
----display.cfg
----klicky.cfg
----macros.cfg
----display.cfg
--KAMP (folder)
----KAMP (folder)
------adaptive_meshing.cfg
------adaptive_purging.cfg
----adaptive_meshing.cfg
----adaptive_purging.cfg
printer.cfg
mainsail.cfg
moonraker.conf
sonar.cfg
crownsnest.conf
KAMP_Settings.cfg
```

This is the .env:

```
github_token=XXX
github_username=XXXX
github_repository=XXX

path_printercfg=/home/pi/printer_data/config/printer.cfg
path_mainsailcfg=/home/pi/mainsail-config/mainsail.cfg
path_moonrakercfg=/home/pi/printer_data/config/moonraker.conf
path_sonarcfg=/home/pi/printer_data/config/sonar.conf
path_crowsnestcfg=/home/pi/printer_data/config/crowsnest.conf
path_kampsettingscfg=/home/pi/printer_data/config/KAMP_Settings.cfg

path_configsfolder=/home/pi/printer_data/config/configs/
path_KAMPfolder=/home/pi/printer_data/config/KAMP/

backup_folder=klipper

```

Staubgeborener, 2023-08-01:
Just curious - could you please try to start from scratch and remove/comment those two files in your `.env`:

```
#path_configsfolder=/home/pi/printer_data/config/configs/
#path_KAMPfolder=/home/pi/printer_data/config/KAMP/
```

RouvenWulff, 2023-08-01:
I've commented out the two lines mentioned. After that, I cleared the repo, then:

```
sudo rm -r klipper
sudo rm -r .git
git init
git add .
git commit -m "create repository"
git branch -M main
git remote add origin https://github.com/RouvenWulff/Voron_Config_Backup.git
git push -u origin main
```

The repo showed the .github/workflows folder, as I have not included this in .gitignore. Then I ran

`~/klipper-backup/script.sh`

which then uploaded these files to my repo:
[image]

Which is confusing as files are being uploaded that are not specified in the .env config. The script is run via cron on startup and after a reboot, this is what I get:

[image]

RouvenWulff, 2023-08-01:
Both. I just cleared the repo again and did not restart the printer after the initial commit as the printer is busy atm.

rolmos, 2023-08-03:
I am in the same situation. All folders are being duplicated and nested into each other.

RouvenWulff, 2023-08-04:
Seems to work for files at least. I uncommented the two directories to be included in the .env file but those however won't be uploaded to my repo. Here's my .env:

```
github_token=XXX
github_username=XXX
github_repository=XXX

path_printercfg=/home/pi/printer_data/config/printer.cfg
path_mainsailcfg=/home/pi/mainsail-config/mainsail.cfg
path_moonrakercfg=/home/pi/printer_data/config/moonraker.conf
path_sonarcfg=/home/pi/printer_data/config/sonar.conf
path_crowsnestcfg=/home/pi/printer_data/config/crowsnest.conf
path_kampsettingscfg=/home/pi/printer_data/config/KAMP_Settings.cfg

path_configsfolder=/home/pi/printer_data/config/configs/
path_KAMPfolder=/home/pi/printer_data/config/KAMP/

backup_folder=klipper
```

and that is what is being uploaded to the repo:

[image]

Staubgeborener, 2023-08-04:
A quick and dirty workaround could be to delete the backup directory as a final last step. So adding something like this at the very end of `script.sh`:

`rm -r "$parent_path/$backup_folder"`

RouvenWulff, 2023-08-04:
What would that be good for? Right now the folders specified in the .env will not be uploaded at all...

Staubgeborener, 2023-08-04:
According to your [post](https://github.com/Staubgeborener/klipper-backup/issues/10#issuecomment-1660687095), the folders will be uploaded (duplicated), if you leave the `cp -r` parameter. The removing part should prevent the duplication.

RouvenWulff, 2023-08-04:
ok. I put back the `-r` in line 23 and added `rm -r "$parent_path/$backup_folder"`to the end of `script.sh`and if I change any file now, it will not be updated in the repo

RouvenWulff, 2023-08-04:
I put back in the `-r` in line 23 of `script.sh`and added the line to delete the backup folder to the end of the script. Whilst not ideal, this works for now.

closed 2023-09-18 by github-actions[bot]