Staubgeborener/Klipper-Backup #11: Instructions unclear on correct "full folder" if you're using anything other than a pi

jonathanalevi, 2023-08-01:
With so many printers coming with other boards (Fly Gemini, BTT Pi, MKS Makerbase), I think the instructions should be updated with the appropriate directories for those boards. 

Personally, I've been unable to get the backup to work at all for any of my printers... 
[image]

Staubgeborener, 2023-08-01:
> With so many printers coming with other boards (Fly Gemini, BTT Pi, MKS Makerbase), I think the instructions should be updated with the appropriate directories for those boards.

There are too many boards and i mean, everyone can pick out their paths for themselves.

> Personally, I've been unable to get the backup to work at all for any of my printers...

Please post your `script.sh` and also your `.env` (without token and any other credentials).

scarteo, 2023-08-15:
you need to add "-r" behind cp in your script.sh file 
like this
`# Copy important files into backup folder
while read -r path; do
  file=$(basename "$path")
  cp -r "$path" "$parent_path/$backup_folder/$file"
done < <(grep 'path_' "$parent_path"/.env | sed 's/^.*=//')
`

closed 2023-09-30 by github-actions[bot]