Staubgeborener/Klipper-Backup #15: "corrupt" error

fushiV, 2023-10-05:
I cant seem to figure out whats causing this "corrupt" error, when I click hard recovery it shows a user name error but my user name is correct in the .env and my files back up just fine. I have tried reinstalling everything from scratch but no change. I also get this other warning while using the backup script from ssh, not sure its related. I also can't figure out why my two of my .cfg don't show up in GitHub with the actual contents but instead just show the file path also pictured below.

[image]
[image]
[image]
[image]
[image]

fushiV, 2023-10-05:
Thank you for the quick reply. Please see the below picture and text of my .env I have tried to keep it simple and just back up the entire printer_data on this install, but previously I had it set to back up each individual cfg file and it had the same problem for the mainsail.cfg (i did not think to look at the timelapse.cfg during the first install).

.env:
---------------
github_token=github_pat_xxxxxxxxxxxxxxxxxx
github_username=fushiv
github_repository=V2-1-backup

#back up whole folder syntax:
path_klipperdata=/home/pi/printer_data/

backup_folder=V2-1_klipper_backup
---------------

[image]

Staubgeborener, 2023-10-05:
Which version of klipper-backup do you use? `cat version`

closed 2023-11-19 by github-actions[bot]