Staubgeborener/Klipper-Backup #16: all goes well until Execute

Heiko910, 2023-10-11:
Hi there,
I am stuck,
I managed to follow the wiki, create a repository, a token, install moonraker support, initialize the Github repository, 
I was going to open the .env file but there was only a .env.example file. I went ahead and created a .env file and used the example as a template.
I configured everything in this new .env file with proper path of all my config files.

Did I mention that I installed it not on a raspberry pi but on a inovato quadra. I corrected all the file path to the correct directories. 

This here is what I am getting when I execute  ~/klipper-backup/script.sh

 ~/klipper-backup/script.sh                                    grep: /home/quadra/klipper-backup/.env: No such file or directory
grep: /home/quadra/klipper-backup/.env: No such file or directory
grep: /home/quadra/klipper-backup/.env: No such file or directory
grep: /home/quadra/klipper-backup/.env: No such file or directory
grep: /home/quadra/klipper-backup/.env: No such file or directory
Reinitialized existing Git repository in /home/quadra/klipper-backup/.git/
WARNING: git-filter-branch has a glut of gotchas generating mangled history
         rewrites.  Hit Ctrl-C before proceeding to abort, then use an
         alternative filtering tool such as 'git filter-repo'
         (https://github.com/newren/git-filter-repo/) instead.  See the
         filter-branch manual page for more details; to squelch this warning,
         set FILTER_BRANCH_SQUELCH_WARNING=1.
Proceeding with filter-branch...

Rewrite 65b48be32ed3d120c90bc2afb890831ed7f7b91b (1/42) (0 seconds passed, remaining 0 predicted)    fatal: /.env: '/.env' is outside repository at '/home/quadra/klipper-backup/.git-rewrite/t'
index filter failed: git rm -r --cached --ignore-unmatch "$parent_path"/.env
On branch main
Your branch is up to date with 'origin/main'.

nothing to commit, working tree clean
Username for 'https://github.com': Heiko910

Any help would be appreciated

Staubgeborener, 2023-10-12:
Could you please show me the output of these two commands:
`ls -la /home/quadra/klipper-backup/`
`cat /home/quadra/klipper-backup/.env` (hide your credentials/token)

Heiko910, 2023-10-12:
I will get you that information when I get home tonight. 
I am usually able to VPN into my home network but last night I turn the printer including the inovato quadra off.

Heiko910, 2023-10-13:
Her is #1 
ls -la /home/quadra/klipper-backup/
total 44
drwxr-xr-x  4 quadra quadra 4096 Oct 10 23:17 .
drwxr-xr-x 41 quadra quadra 4096 Oct 12 00:17 ..
-rw-r--r--  1 quadra quadra 1327 Oct 10 23:16 .env
-rw-r--r--  1 quadra quadra  980 Oct 10 23:17 .env.example
drwxr-xr-x  8 quadra quadra 4096 Oct 12 22:13 .git
drwxr-xr-x  3 quadra quadra 4096 Oct 10 23:15 .github
-rw-r--r--  1 quadra quadra   17 Oct 10 23:15 .gitignore
-rw-r--r--  1 quadra quadra 2164 Oct 10 23:15 install.sh
-rw-r--r--  1 quadra quadra  432 Oct 10 23:15 README.md
-rwxr-xr-x  1 quadra quadra 1251 Oct 10 23:15 script.sh
-rw-r--r--  1 quadra quadra    5 Oct 10 23:15 version

#2 cat /home/quadra/klipper-backup/.env
github_token=gh************************************************
github_username=H**********
github_repository=SV06-Klipper-Backup

path_printercfg=/home/quadra/printer_data/config/printer.cfg
path_adxlmcucfg=/home/quadra/printer_data/config/adxlmcu.cfg
path_moonrakercfg=/home/quadra/printer_data/config/moonraker.conf
path_osskccfg=/home/quadra/printer_data/config/osskc.cfg
path_CALIBRATIONcfg=/home/quadra/printer_data/config/cfgs/CALIBRATION.cfg
path_crowsnestcfg=/home/quadra/printer_data/config/crowsnest.conf
path_MECHANICAL_GANTRY_CALIBRATIONcfg=/home/quadra/printer_data/config/cfgs/MECHANICAL_GANTRY_CALIBRATION.cfg
path_misc-macroscfg=/home/quadra/printer_data/config/cfgs/misc-macros.cfg
path_PARKINGcfg=/home/quadra/printer_data/config/cfgs/PARKING.cfg
path_TEST_SPEEDcfg=/home/quadra/printer_data/config/cfgs/TEST_SPEED.cfg
path_Adaptive_Meshingcfg=/home/quadra/printer_data/config/cfgs/kamp/Adaptive_Meshing.cfg
path_CALIBRATIONcfg=/home/quadra/printer_data/config/cfgs/kamp/KAMP_Settings.cfg
path_Line_Purgecfg=/home/quadra/printer_data/config/cfgs/kamp/Line_Purge.cfg
path_Smart_Parkcfg=/home/quadra/printer_data/config/cfgs/kamp/Smart_Park.cfg
path_Voron_Purgecfg=/home/quadra/printer_data/config/cfgs/kamp/Voron_Purge.cfg

#back up whole folder syntax:
#path_klipperdata=/home/quadra/printer_data/

backup_folder=klipper

and here is #3 
0.3.6

BTW I think it is now all working. A reboot of the quadra machine and a restart of klipper and the update function in klipper seems to be working.
I checked and all the config files that I configured in the .env file are now actually being backed up.

As i mentioned initially there was no .env file but a .env.example. I just edited it and renamed it to just .env. I hope that is what needed to happen.

Please let me know if you see anything that is out of the ordinarry.
Thanks Heiko

Staubgeborener, 2023-10-13:
Good to hear. Your `.env` file looks correct for me. 

But weird that you don't get a `.env` file. Normally, the [installation process](https://github.com/Staubgeborener/klipper-backup/blob/main/install.sh#L34) should create one. As a one last step (just to handle this `.env` problem), could explain exactly step by step how you have done this? Please don't answer with "i followed the wiki". For example you can simply give me the relevant output of `history`.

Btw, please use the [markdown syntax](https://docs.github.com/de/get-started/writing-on-github/getting-started-with-writing-and-formatting-on-github/basic-writing-and-formatting-syntax) for easier readability. You can put your codes into \` like `this` or with hightling syntax
```shell
#this is
"my"
code
```

Heiko910, 2023-10-13:
I don't know where to find the relevant history.
This was my first time doing anything on a GitHub account and I literally
followed your installation procedure. The changes to the configuration from
pi to quadra were all after the fact.
The markdown syntax is also completely new to me. I would have to read up
on that first
Sorry, nooby here.

Heiko910, 2023-10-13:
#this is
"ls -la /home/quadra/klipper-backup/
total 44
drwxr-xr-x 4 quadra quadra 4096 Oct 10 23:17 .
drwxr-xr-x 41 quadra quadra 4096 Oct 12 00:17 ..
-rw-r--r-- 1 quadra quadra 1327 Oct 10 23:16 .env
-rw-r--r-- 1 quadra quadra 980 Oct 10 23:17 .env.example
drwxr-xr-x 8 quadra quadra 4096 Oct 12 22:13 .git
drwxr-xr-x 3 quadra quadra 4096 Oct 10 23:15 .github
-rw-r--r-- 1 quadra quadra 17 Oct 10 23:15 .gitignore
-rw-r--r-- 1 quadra quadra 2164 Oct 10 23:15 install.sh
-rw-r--r-- 1 quadra quadra 432 Oct 10 23:15 README.md
-rwxr-xr-x 1 quadra quadra 1251 Oct 10 23:15 script.sh
-rw-r--r-- 1 quadra quadra 5 Oct 10 23:15 version"
code

#this is
"cat /home/quadra/klipper-backup/.env
github_token=gh************************************************
github_username=H**********
github_repository=SV06-Klipper-Backup

path_printercfg=/home/quadra/printer_data/config/printer.cfg
path_adxlmcucfg=/home/quadra/printer_data/config/adxlmcu.cfg
path_moonrakercfg=/home/quadra/printer_data/config/moonraker.conf
path_osskccfg=/home/quadra/printer_data/config/osskc.cfg
path_CALIBRATIONcfg=/home/quadra/printer_data/config/cfgs/CALIBRATION.cfg
path_crowsnestcfg=/home/quadra/printer_data/config/crowsnest.conf
path_MECHANICAL_GANTRY_CALIBRATIONcfg=/home/quadra/printer_data/config/cfgs/MECHANICAL_GANTRY_CALIBRATION.cfg
path_misc-macroscfg=/home/quadra/printer_data/config/cfgs/misc-macros.cfg
path_PARKINGcfg=/home/quadra/printer_data/config/cfgs/PARKING.cfg
path_TEST_SPEEDcfg=/home/quadra/printer_data/config/cfgs/TEST_SPEED.cfg
path_Adaptive_Meshingcfg=/home/quadra/printer_data/config/cfgs/kamp/Adaptive_Meshing.cfg
path_CALIBRATIONcfg=/home/quadra/printer_data/config/cfgs/kamp/KAMP_Settings.cfg
path_Line_Purgecfg=/home/quadra/printer_data/config/cfgs/kamp/Line_Purge.cfg
path_Smart_Parkcfg=/home/quadra/printer_data/config/cfgs/kamp/Smart_Park.cfg
path_Voron_Purgecfg=/home/quadra/printer_data/config/cfgs/kamp/Voron_Purge.cfg

#back up whole folder syntax:
#path_klipperdata=/home/quadra/printer_data/

backup_folder=klipper"
code

Staubgeborener, 2023-10-13:
Just type `history` in your terminal. This will output your last entered commands from the current user. You should see all commands from the installation procedure.

Heiko910, 2023-10-14:
Here is the history with all my typos. 

_1  sync
    2  sudo apt update
    3  passwd
    4  sudo apt update
    5  sudo apt upgrade
    6  sudo apt autoremove
    7  sudo apt-get update && sudo apt-get install git -y
    8  cd ~ && git clone https://github.com/dw-0/kiauh.git
    9  ./kiauh/kiauh.sh
   10  ls -l /dev/serial/by-path/
   11  ls -l /dev/serial/by-id/
   12  ./kiauh/kiauh.sh
   13  ls -l /dev/serial/by-path/
   14  cd ~/moonraker/scripts
   15  ./set-policykit-rules.sh
   16  sudo service moonraker restart
   17  cd ~/moonraker/scripts
   18  ./install-moonraker.sh
   19  sudo apt update
   20  apt list --upgradable
   21  sudo apt upgrade
   22  ./kiauh/kiauh.sh
   23  cd ~/crowsnest
   24  sudo make install
   25  ./kiauh/kiauh.sh
   26  ls -l /dev/serial/by-path/
   27  sudo apt update
   28  sudo apt upgrade
   29  ./kiauh/kiauh.sh
   30  ls /dev/v41/by-id
   31  dir
   32  cd dev
   33  ls /dev/v41/by-id
   34  ls /dev/video/by-id
   35  cd ~/crowsnest
   36  tools/dev-helper .sh -c
   37  tools/dev-helper.sh -c
   38  ls /dev/v4l/by-id
   39  ls /dev/v4l/by-path
   40  ls /dev/v4l/by-id
   41  tools/dev-helper.sh -c
   42  ls /dev/video/by-id
   43  ls /dev/v4l/by-id
   44  cd
   45  sudo usermod -aG video quadra
   46  ~/klippy-env/bin/pip install -v numpy
   47  sudo apt update
   48  sudo apt upgrade
   49  sudo apt install python3-numpy python3-matplotlib libatlas-base-dev
   50  cd klipper
   51  make clean
   52  make menuconfig
   53  make
   54  ls /dev/
   55  cd
   56  ls /dev/
   57  cd klipper
   58  ls /dev/
   59  sudo mount /dev/sda1 /mnt
   60  ls /dev/
   61  make menuconfig
   62  ls /dev/
   63  make menuconfig
   64  make
   65  ls /dev/
   66  sudo mount /dev/sda1 /mnt
   67  sudo cp out/klipper.uf2 /mnt
   68  sudo umount /mnt
[... 167 lines cut ...]
  236  Rewrite 65b48be32ed3d120c90bc2afb890831ed7f7b91b (1/42) (0 seconds passed                         , remaining 0 predicted)    fatal: /.env: '/.env' is outside repository at '/hom                         e/quadra/klipper-backup/.git-rewrite/t'
  237  index filter failed: git rm -r --cached --ignore-unmatch "$parent_path"/.                         env
  238  On branch main
  239  Your branch is up to date with 'origin/main'.
  240  ~/klipper-backup/script.sh
  241  sudo apt update
  242  sudo apt upgrade
  243  sudo apt-get install python3-httpx
  244  sudo apt autoremove
  245  sudo apt update
  246  apt list --upgradable
  247  sudo apt upgrade
  248  sudo apt update
  249  sudo apt-get install python3-httpx
  250  cd ~/moonraker/scripts
  251  ./set-policykit-rules.sh
  252  sudo service moonraker restart
  253  cd
  254  sudo apt update
  255  suo apt upgrade
  256  sudo apt upgrade
  257  ls -la /home/quadra/klipper-backup/
  258  cat /home/quadra/klipper-backup/.env
  259  cat /home/quadra/klipper-backup/version
  260  history__

Staubgeborener, 2023-10-15:
This is why I said "_[...] the relevant output [...]_". Maybe you want to consider **deleting** this post as you leaked your token, private email, etc. GitHub also does make a "post history", so you can also see after editing what the user posted/edited (your token for example).

So directly change the token I would say.

[image]

Heiko910, 2023-10-15:
I already changed my access token. 
But, yes, I guess this just above my head.
It isn't working. I followed the last step in the wiki to execute the script and to add the macro to be able execute the script from within mainsail but now I get another error message in mainsail.

closed 2023-10-15 by Heiko910

Staubgeborener, 2023-10-15:
The message in mainsail regarding updates corresponds to [this](https://github.com/Staubgeborener/klipper-backup/issues/6) issue. TLTR: You should be able to use this script for backup purpose, but you're not able to update it.

This is why i added a small hint in the wiki:
[image]

Heiko910, 2023-10-16:
how do I reverse this install?

Staubgeborener, 2023-10-16:
There is nothing which can be "reversed". This scripts creates a folder and that's it, there are no binaries which will be installed. 

`sudo rm -r ~/klipper-backup`

And maybe delete the added lines in your `moonraker.conf`. That's pretty much it.

Staubgeborener, 2023-10-16:
> one, followed the instruction of YouTuber that used your script 

Now I'm interested, could you please post the link to this video. Maybe I can use it for the wiki or readme file if the quality is good enough.