Staubgeborener/Klipper-Backup #18: Updates rejected...remote contains work that not local

DrFate09, 2023-11-12:
Trying to set up this plugin and when I try and push my files to my GitHub repo, I am faced with this error:

hint: Updates were rejected because the remote contains work that you do
hint: not have locally. This is usually caused by another repository pushing
hint: to the same ref. You may want to first integrate the remote changes
hint: (e.g., 'git pull ...') before pushing again.
hint: See the 'Note about fast-forwards' in 'git push --help' for details.

Any thoughts on what I am doing wrong?

DrFate09, 2023-11-13:
UPDATE: I used git push --force to make it force the files to my repo.

Staubgeborener, 2023-11-14:
Sadly i can't find your backup repository to check your commit history (which would help to give you an exact statement and not to guess) but glad you find it out by yourself. Exactly as it says: I think you had files in your repository which did not exist in your folder.

closed 2023-11-14 by Staubgeborener