Staubgeborener/Klipper-Backup #23: Need to restore my configs to a "recovered" printer, not familiar enough with Git

ccatlett1984, 2023-12-30:
Could you make a quick writeup on how to clone the existing repo down to a new install, and what steps would then be required to have it start pushing changes up to the repo.

My SD card died, awesome that I have my configs backed up. Ideally I'd like to be able to "pull" the existing configs down, without manual copying.

Staubgeborener, 2023-12-30:
You can simply use `git clone url-to-your-repository` and run the `./script.sh` file inside this directory. Or am I missing anything?

ccatlett1984, 2023-12-30:
Scenario: SD Card died
NEW SD Card imaged and can ssh into. Installed klipper/etc. using KIAUH
want to be able to run a command and pull down all the configs from my private repo.

Staubgeborener, 2023-12-31:
tbh never thought about that as i would simply use some `cp` and/or `cp -r` commands to copy all the files to the paths defined in `.env` file. So you would need around a minute to restore everything by hand.

```shell
#!/bin/bash

backup_folder=$(grep "backup_folder" .env | awk -F'=' '{print $2}')
echo $backup_file
while IFS= read -r line; do
    if [[ $line == path_* ]]; then
        destination_path=$(echo "$line" | awk -F'=' '{print $2}')
        backup_file=$(echo "$line" | awk -F'/' '{print $NF}')
        cp "$backup_folder/$backup_file" "$destination_path"
    fi
done < .env

echo "Restored files"
```

I have just written a script that fulfills your requirements. However, I have only tested it with some test data on a non-klipper system and not in a live environment, so no guarantee.

The script **must** be located in the same directory as the `.env`.

So run `git clone [...]` on your own repository, copy this script e.g. as `restore.sh` into the folder and then run it with a 
```shell
chmod +x ./restore.sh
./restore.sh
```

Do not be confused by any error messages. Execute the script, **check all directories manually** (!) with some `ls` commands and please report if it worked. If not, please tell me **in detail** what did not work.

Staubgeborener, 2024-01-18:
No feedback after 3 weeks, so I'll close this as I assume that the "problem" has been solved

closed 2024-01-18 by Staubgeborener