Staubgeborener/Klipper-Backup #27: Backup not including subfolders

muhquh, 2024-01-12:
Backup is not including subfolders:

`cp: -r not specified; omitting directory '/home/pi/printer_data/config/macros'`

got a workaround by changing

   ` cp $file $HOME/$backup_folder/`
in the sript.sh to

      `cp -r $file $HOME/$backup_folder/`

is there an other way to do it?

Staubgeborener, 2024-01-12:
I've looked at it and tested it - you're right. I've just [corrected](https://github.com/Staubgeborener/klipper-backup/commit/1e504065eb36db5b60d21b22b09fa81f9aa2d8ab#diff-116b4e2bac3e35befdfae3f3b6eb88911689c30bfe6696d9e6b0139da8cb5e72R31) it.

(At the same time, I also found a [security issue](https://github.com/Staubgeborener/klipper-backup/commit/1e504065eb36db5b60d21b22b09fa81f9aa2d8ab#diff-116b4e2bac3e35befdfae3f3b6eb88911689c30bfe6696d9e6b0139da8cb5e72R38) with the .gitignore file - this was completely ignored in the user repository)

Staubgeborener, 2024-01-13:
**One step back!** @muhquh 

I have replaced `cp -r` with `cp` again.

Reason: When using cp -r, push errors occur in certain scenarios, all of which should be covered.

Much more relevant, however, is that the user loses control over what is pushed. Let's assume you have a path `printer_data/config/` and in it are all kinds of folders that you want to push. But there are also numerous folders and files in it that should not be pushed, for example `printer_data/config/mysecrets`, then they will all be pushed anyway. The user would be forced to cover all variants in the [.gitignore](https://github.com/Staubgeborener/klipper-backup/blob/main/.gitignore), which complicates everything. If you now want to push the folder `printer_data/config/` and `printer_data/config/1/` as well as `printer_data/config/2/`, then please specify these folders separately in the [.env](https://github.com/Staubgeborener/klipper-backup/blob/main/.env.example) (see [this](https://github.com/Staubgeborener/klipper-backup/wiki/Installation%3A-Configuration#env)) as follows:

```ini
path_klipperdata=printer_data/config/*
path_klipperdata=printer_data/config/1/*
path_klipperdata=printer_data/config/2/*
```

Just pay attention to the wildcard `*`.

This gives the user much more control over what exactly should be pushed. I hope the explanation of why I am taking this step was understandable.

muhquh, 2024-01-13:
Got you. Thanks😃

closed 2024-01-13 by muhquh

Staubgeborener, 2024-01-13:
Alright, I also added a new [Q&A](https://github.com/Staubgeborener/klipper-backup/wiki/Questions-and-Answers) section with [this information](https://github.com/Staubgeborener/klipper-backup/wiki/Questions-and-Answers#question-i-want-to-push-folders-recursively-with-subfolders-how-does-it-work).
(I have the feeling that the wiki needs to be tidied up, and maybe we need a proper doc website at this point...)