Staubgeborener/Klipper-Backup #32: "Klipper-backup is not up to date" Message when running from Mainsail

zer010gic, 2024-01-19:
When I run the macro to execute the Klipper-backup script from Mainsail I am getting a message that Klipper-backup is not up to date.  However if I run the script from the terminal it states that it is up to date.  I am also using Mainsail update manager to update the script and I am currently running 1.1.0-0-g713d4cfe.

[image]
[image]
[image]

Staubgeborener, 2024-01-19:
Thanks. The problem here is that when running in mainsail, things like `git ls-remote` can't work at all, because you have to be in the klipper-backup directory at that time. The same error message can be achieved by running something like `./klipper-backup/script.sh` in the `$home` directory.

Long story short: Ignore this message for now.

Since I cannot find an elegant way to fix this, I will simply remove the corresponding line completely in the next time.  Maybe I start again with something like a `version` file to check this, like I did a couple of weeks ago.

Staubgeborener, 2024-01-19:
I think i found a good solution with `git -C [...]`, could you please test this with the mainsail ui?

zer010gic, 2024-01-20:
The latest update fixed it.

closed 2024-01-20 by zer010gic