Staubgeborener/Klipper-Backup #33: changes made to .env file don't seem to get applied?

slovenian6474, 2024-01-20:
After following the instructions, everything worked fine.  I wanted to change what repository it was going to. I changed that in the .env file but the command seems to still be using the original repository even though the .env no longer show it. I've tried removing the entire "klipper-backup" folder and all it's contents, starting over and reinstalling but somehow it still only runs what was originally in the .env the first time it ran.

Staubgeborener, 2024-01-20:
Well, yeah, you are completely right. The reason is that the script accesses the `.env` variable in this case, but `git` itself "stores it" and then when your folder/repository is accessed, it pulls it from there.

The information is stored in the `config_backup` folder:

```shell
[user@klipper .git]$ cat ~/config_backup/.git/config
[init]
        defaultBranch = main
[core]
        repositoryformatversion = 0
        filemode = true
        bare = false
        logallrefupdates = true
[user]
        name = user
        email = user@klipper
[remote "origin"]
        url = https://github_pat_supersecrettoken@github.com/staubgeborener/myklipperbackuprepository.git
        fetch = +refs/heads/*:refs/remotes/origin/*
[advice]
        skippedCherryPicks = false
[branch "main"]
        remote = origin
        merge = refs/heads/main
```

Two solutions: 
1. Also delete the `config_backup` folder `sudo rm -r ~/config_backup` or 
2. Edit the line in `[remote "origin"]` section from this
`url = https://github_pat_supersecrettoken@github.com/staubgeborener/myklipperbackuprepository.git`
to this
`url = https://github_pat_supersecrettoken@github.com/staubgeborener/mynewrepository.git`

If you only want to change the repository and keep your commit history, I recommend step two. When the script is executed again, your new repository will look like your previous one.

slovenian6474, 2024-01-20:
Excellent!  I did #2 and worked just fine!  Thanks for the quick response!

Staubgeborener, 2024-01-21:
Nice! I have also [adapted the script](https://github.com/Staubgeborener/klipper-backup/commit/b6c17772c2e7e7c77c2a26f734fd79f01dacb2cf), in the future you will not have to delete folders or edit the `.git/config` manually. The script now checks whether the repository or the branch_name in the `.env` have been changed and adjusts everything automatically.

closed 2024-01-21 by Staubgeborener