Staubgeborener/Klipper-Backup #34: Commit username

Crankyyankee713, 2024-01-21:
Let me start by saying that I have very little github experience but many of us are just using our klipperized 3d printers and want a nice backup solution so we are using this. 

Currently the latest update on 1/12 fixed my issue that i had where it wouldn't backup at all. I received 
fatal: could not read password for  'https://ghp_xxxxxxxxxxx@github.com' : no such device or address and would not work at all.

Now the backup works fine with the newest update but all the commits show Mahallon as the user.
I attempted to edit the comit username in the .env file but nothing seems to change.

Staubgeborener, 2024-01-21:
> Let me start by saying that I have very little github experience but many of us are just using our klipperized 3d printers and want a nice backup solution so we are using this.

The aim was also to make this as simple as possible. In the best case, out of the box, but users still have to define some variables.

> Now the backup works fine with the newest update but all the commits show Mahallon as the user. I attempted to edit the comit username in the .env file but nothing seems to change.

So you edited [commit_username=""](https://github.com/Staubgeborener/klipper-backup/blob/main/.env.example#L5) to something like `commit_username="Crankyyankee713"`?

I have just tested it, and it works for me without any problems. First, I just had it as `commit_username=""`, so my `whoami` was used for pushing when calling `./script.sh` (which should be Mahallon in your case i guess). You can also see this by running `whoami` in your terminal:
```shell
[user@klipper klipper-backup]$ whoami
user
```
Then I adjusted the `.env` as follows `commit_username="klipper-backup-user"` and ran `./script.sh` again. Now, `klipper-backup-user` appears as username in the commit history.

Just test this again and follow this. Step by step. If you're 100% sure this is not the problem, please provide the following information/output:

1.
```shell
cd ~/config_backup
git config user.name
```

2.
```shell
whoami
```

3. Content of your `.env` file (without the token!):
```shell
cat ~/klipper-backup/.env
```

Crankyyankee713, 2024-01-21:
Yes, I tried difference names in the defined section there, ran script in terminal, ran it in main sail. Edited some files to make sure it was pushing new ones. Everything shows the commits by that name and i wasnt sure why.  First i left that empty and expected to see pi as the user. 

Here are the outputs
[image]
[image]
[image]

Staubgeborener, 2024-01-21:
Now I'm quite confused. Is [this](https://github.com/Mahallon) one of your GitHub profiles? Maybe Mahallon is in your global git config? Can you please run `git config --list` in `klipper-backup` and also in `~`.

Crankyyankee713, 2024-01-21:
I found that profile as well but no its not mine. I created my github last year and only used for backups, I made a new repo when attempting to follow your guide. So my understanding appears to be right and we both do not know how it displays that name. It does not appear to be an issue within the pi itself at this point. 

pi@raspberrypi:~/klipper-backup $ git config --list
user.name=Crankyyankee713
core.repositoryformatversion=0
core.filemode=true
core.bare=false
core.logallrefupdates=true
remote.origin.url=https://github.com/Staubgeborener/klipper-backup.git
remote.origin.fetch=+refs/heads/*:refs/remotes/origin/*
branch.main.remote=origin
branch.main.merge=refs/heads/main
moonraker.instance=moonraker@13afffc1acd445b78ee465e82b895551
pi@raspberrypi:~/klipper-backup $ cd
pi@raspberrypi:~ $ cat ~/config_backup/.git/config
[init]
        defaultBranch = main
[core]
        repositoryformatversion = 0
        filemode = true
        bare = false
        logallrefupdates = true
[user]
        name = Crankyyankee713
        email = pi@raspberrypi
[remote "origin"]
        url = https://ghp_xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx@github.com/Crankyyankee713/Voron-2.4-Config-Backup.git
        fetch = +refs/heads/*:refs/remotes/origin/*
[advice]
        skippedCherryPicks = false
[branch "main"]
        remote = origin
        merge = refs/heads/main

Staubgeborener, 2024-01-21:
> It does not appear to be an issue within the pi itself at this point.

It seems to be an issue with your system and not klipper-backup itself.

In your `cat ~/config_backup/.git/config` output is this entry: `email = pi@raspberrypi`

Please edit the `commit_email=""` parameter in `.env` with an email to overwrite this global setting just for the klipper-backup script (for global settings aka for every git usecase on your system: this one can also be changed to your own email for commits with `git config --global user. email "yourownunique@email.com"`).

@Tylerjet just made a short test with this email.

[image]

Your problem seems to be related with this article: https://docs.github.com/en/enterprise-cloud@latest/pull-requests/committing-changes-to-your-project/troubleshooting-commits/why-are-my-commits-linked-to-the-wrong-user#

Crankyyankee713, 2024-01-21:
Thanks for the help, I did some searching but didn't come across that. That was the default I suppose from the raspberry pi OS image. Everything looks good now!  Might be worth noting for the rookies to fill in the optional commit username/email

closed 2024-01-21 by Crankyyankee713

Tylerjet, 2024-01-21:
Yeah we just chatted about it the plan is to make it use a more unique string (probably the commit hash of the most recent commit) instead of just using the hostname. That way the email is always unique unless set specifically in .env.

Staubgeborener, 2024-01-22:
Problem is now fix'd with latest commit to prevent issues like this.