Staubgeborener/Klipper-Backup #36: Unable to push backup

CallumRD1, 2024-01-23:
I followed the installation instructions and it pushed to git once but on every subsequent push I get this error:

`Klipper-backup is up to date

Skipping symbolic link: printer_data/config/K-ShakeTune
Skipping symbolic link: printer_data/config/mainsail.cfg
Skipping symbolic link: printer_data/config/print_area_bed_mesh.cfg
Skipping file: printer_data/config/printer-20240121_185359.cfg
Skipping symbolic link: printer_data/config/timelapse.cfg
New branch in .env detected, rename  to main branch
fatal: Invalid branch name: 'HEAD'
interactive rebase in progress; onto 5082fce
Last command done (1 command done):
   pick 786037a New backup from 01-23-24
No commands remaining.
You are currently editing a commit while rebasing branch 'main' on '5082fce'.
  (use "git commit --amend" to amend the current commit)
  (use "git rebase --continue" once you are satisfied with your changes)

nothing to commit, working tree clean
From https://[REDACTED]@github.com/[REDACTED].git
From https://github.com/[REDACTED]
 * branch            main       -> FETCH_HEAD
fatal: It seems that there is already a rebase-merge directory, and
I wonder if you are in the middle of another rebase.  If that is the
case, please try
	git rebase (--continue | --abort | --skip)
If that is not the case, please
	rm -fr ".git/rebase-merge"
and run me again.  I am stopping in case you still have something
valuable there.

Branch 'main' set up to track remote branch 'main' from 'origin'.
Everything up-to-date`

None of the commends suggested in the error message appear to do anything. The rebase commands give the response: "fatal: No rebase in progress?".

Staubgeborener, 2024-01-23:
No idea what you did wrong, but the `git rebase` has to be done in the `config_backup` folder which you can reach with `cd ~/config_backup`

CallumRD1, 2024-01-23:
I couldn't figure out all of what you're talking about above, so I decided to just create a new repo and use that as a clean target. I pointed the .env file to it and ran your program. It uploaded once, but for some reason the commit date is 3 hours in the past, and it's not pushing the latest versions of the files. For subsequent pushes, I get this code out, and the versions of my printer.cfg are still incorrect. They don't include text changes I put in hours ago.

`~/klipper-backup/script.sh "my commit message2"
NEW klipper-backup version available!

Skipping symbolic link: printer_data/config/K-ShakeTune
Skipping symbolic link: printer_data/config/mainsail.cfg
Skipping symbolic link: printer_data/config/print_area_bed_mesh.cfg
Skipping file: printer_data/config/printer-20240121_185359.cfg
Skipping symbolic link: printer_data/config/timelapse.cfg
New branch in .env detected, rename  to main branch
fatal: Invalid branch name: 'HEAD'
Not currently on any branch.
nothing to commit, working tree clean
From https://[snip]@github.com/[snip]-2.git
From https://github.com/[snip]-2
 * branch            main       -> FETCH_HEAD
HEAD is up to date.
Branch 'main' set up to track remote branch 'main' from 'origin'.
Everything up-to-date
`

Tylerjet, 2024-01-24:
When you created the new repo did you also delete the config_backup folder, as if you did not it will have all existing commit history and setup.

The rebase error i have found comes from using an existing repo or from pushing to the same repo from another location then pushing changes again from a different location that has conflicting changes and causes a rebase conflict. I'm working on a PR that should be able to self resolve this.

CallumRD1, 2024-01-24:
I did not delete the config_backup folder; I wasn't aware that there was any need to.

Thank you for the help with this. Once you've implemented the fix you mentioned, could you explain to me how I should clean up my existing config so I can try this again properly?

Tylerjet, 2024-01-24:
Yeah usually when switching to a new backup repo i have found its easiest to wipe on the config backup so that it will create all new .git/config files.

The hope with the PR will be that no changes will be needed as it will just see there are rebase conflicts and perform the steps to resolve said conflicts.

Tylerjet, 2024-01-24:
@CallumRD1 could you try the PR i just made.

```git pull origin pull/37/head```

CallumRD1, 2024-01-24:
How do I use that command? I tried running it in the config_backup folder and got this: 
[image]

Tylerjet, 2024-01-24:
run that in the klipper-backup folder

CallumRD1, 2024-01-25:
That appears to have done it. I ran that command in the Klipper backup folder, ran your script, it failed, so I deleted the config_backup folder as you recommended above, and now it's working. I'm seeing changes to the files propagate. Now I just need to automate it.

CallumRD1, 2024-01-25:
I used your chron entry but it's not pushing any changes upon a reboot. I'm only getting changes to push to get when I manually call the script.

Tylerjet, 2024-01-25:
Which cron entry?

CallumRD1, 2024-01-26:
[image]
I'm using your code from https://github.com/Staubgeborener/klipper-backup/wiki/execute but it's not pushing to my repo. I've rebooted the system about a dozen times and it hasn't pushed anything. When I run the code manually it works.

Tylerjet, 2024-01-26:
could you try ```@reboot sleep 60 && $HOME/klipper-backup/script.sh```

was able to get the cron to redirect the scripts output and it looks like the network is not ready when @reboot scripts are ran as i see ```fatal: Could not resolve host: github.com``` so it needs to be delayed so the network has time to start.

Tylerjet, 2024-01-26:
that will sleep for 1 minute before running the command

Tylerjet, 2024-01-26:
I will look at possibly making documentation for creating a service instead, as services can run on startup but you can also specify that the service should wait for the network be be available before attempting to run.

CallumRD1, 2024-01-26:
The delayed start worked perfectly! Thank you!

Tylerjet, 2024-01-26:
Sweet! If there are no other issues feel free to close this issue. 😄

Staubgeborener, 2024-01-26:
> I will look at possibly making documentation for creating a service instead

Yeah, it's time for a [real](https://github.com/Staubgeborener/klipper-backup/tree/docs) documentation.

PR #37 (2024-01-25): Reordered git actions and order of operations
Script will now pull any changes to the local branch first (should prevent any need for rebase or merge), then copy the backup to the repository folder, add and commit the backup's files, then push it back to the remote repository.

Should resolve #36 as well

Please test on your end too, I did very small tests as the only main changes was the order in which some commands were ran with a few additional checks as well.
closed 2024-01-26 by Staubgeborener