Staubgeborener/Klipper-Backup #48: [Feature request]:  Can Add something in the Automated Backups Doc?

GHTGXL, 2024-02-03:
### Is your feature request related to a problem? Please describe.

Because I'm using CB1, the link is not /home/mks/klipper-backup/script.sh, it is /home/biqu/klipper-backup/script.sh, and the user is also different, is not root, is biqu.

### Describe the solution you'd like

So can add the warning about the different link and the username?

### Describe alternatives you've considered

Or another way to remind you about these two?

Staubgeborener, 2024-02-03:
Are we talking about the [systemd service file](https://staubgeborener.github.io/klipper-backup/)?

Tylerjet, 2024-02-03:
Will be updated in the docs please change ExecStart line in the service to:

```ExecStart=/bin/bash -c 'bash $HOME/klipper-backup/script.sh "New Backup on boot $(date +%%D)"'```

Staubgeborener, 2024-02-04:
The docs are updated now

closed 2024-02-04 by Staubgeborener