Staubgeborener/Klipper-Backup #49: [Bug]: fail to start klipper backup service

badTastingBread, 2024-02-03:
### What happened

When running `sudo systemctl start klipper-backup.service`, I do get the error
`● klipper-backup.service - Klipper Backup Service
     Loaded: loaded (/etc/systemd/system/klipper-backup.service; enabled; vendor preset: enabled)
     Active: failed (Result: exit-code) since Sat 2024-02-03 19:37:40 CET; 15s ago
    Process: 2523 ExecStart=/bin/bash -c /home/mks/klipper-backup/script.sh "New Backup on boot $(date +%D)" (code=exited, status=217/USER)
   Main PID: 2523 (code=exited, status=217/USER)
        CPU: 0
`

Running `journalctl -xe
` I do get

`The unit klipper-backup.service has entered the 'failed' state with result 'exit-code'.
Feb 03 19:35:02 mainsail systemd[1]: Failed to start Klipper Backup Service.
░░ Subject: A start job for unit klipper-backup.service has failed
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░
░░ A start job for unit klipper-backup.service has finished with a failure.
░░
░░ The job identifier is 1117 and the job result is failed.
Feb 03 19:35:02 mainsail sudo[2480]: pam_unix(sudo:session): session closed for user root
Feb 03 19:36:23 mainsail python[690]: [websockets.py:on_close()] - Websocket Closed: ID: 3888790688 Close Code: 1001, Close Reason: None, Pong Time Elapsed: 6.22
Feb 03 19:37:40 mainsail sudo[2520]:     arne : TTY=pts/1 ; PWD=/home/arne ; USER=root ; COMMAND=/usr/bin/systemctl start klipper-backup.service
Feb 03 19:37:40 mainsail sudo[2520]: pam_unix(sudo:session): session opened for user root(uid=0) by arne(uid=1000)
Feb 03 19:37:40 mainsail systemd[2523]: klipper-backup.service: Failed to determine user credentials: No such process
Feb 03 19:37:40 mainsail systemd[2523]: klipper-backup.service: Failed at step USER spawning /bin/bash: No such process
░░ Subject: Process /bin/bash could not be executed
` 

### What did you expect to happen

Expected it to starts

### How to reproduce

Installed like described in the wiki into the regular home folder on a mainsail system on a raspberry pi 4b 

### Additional information

I hope I didnt bother you too much, but I dont know what to do anymore. 

Regards Arne :)

Staubgeborener, 2024-02-03:
Please run `whoami` and post the output here.

Tylerjet, 2024-02-03:
This will be updated in the docs please change ExecStart line in the service to:

```ExecStart=/bin/bash -c 'bash $HOME/klipper-backup/script.sh "New Backup on boot $(date +%%D)"'```

Tylerjet, 2024-02-03:
Oh wait yours is having an issue as well with /bin/bash what OS are you running on the RPI as it may be /bin/bash is not used and it might need to be /bin/sh or similar.

badTastingBread, 2024-02-04:
spelled my username wrong, as Staubgeborener said.

Thank you :)

closed 2024-02-04 by badTastingBread

Staubgeborener, 2024-02-04:
We also fix this issue completely by updating the automated backup docs section. It's now flexible as it can be.