Staubgeborener/Klipper-Backup #51: [Bug]: Unable to push to repo due to compressing error

seacred, 2024-02-06:
### What happened

After running ./script.sh the compressing procedure takes a while and the upload fails.
[image]

### What did you expect to happen

Pushing configfiles to repository

### How to reproduce

unable to tell

Staubgeborener, 2024-02-06:
What exactly are you trying to upload? Is the amount of data perhaps a bit huge? Sounds like it's related to something like [this](https://stackoverflow.com/questions/12651749/git-push-fails-rpc-failed-result-22-http-code-411) on your system.

Could you please check the `postBuffer` with

```shell
cd ~/config_backup
git config --get http.postBuffer
```

You could try to change the value with `git config http.postBuffer 524288000` and try to run `./script.sh` again. If it still fails, please show you `.env` file (but hide your token).

seacred, 2024-02-06:
None of my Files is bigger then 400kb. 
Running the command does not generate any output.
My .ev is nothing special and mostly default and i've checked the username and repo.
[image]

It works until like 87% during the compression and tries to push after reaching to 100% but fails with error 400.

Staubgeborener, 2024-02-06:
Odd. Only thing I can tell you: this is not a Klipper-Backup error but an issue with your system.

I would do the following:

```shell
# delete and re create your GitHub repository form scratch
sudo apt install --only-upgrade git
# or maybe updating the whole system with sudo apt update && sudo apt upgrade
sudo rm -rf ~/config_backup
```
After this run `./script.sh`.

CallumRD1, 2024-02-17:
I had klipper backup working fine for a while and then it stopped pushing backups altogether. After I realized this I manually ran the script so I'd get the verbose output and found similar set of errors. The compression step takes about 30 minutes and then it fails to push. It's just trying to backup my config folder, but when I comment out that folder in the .env and just try to have it backup the printer.cfg single file I get the same error.

[image]

Tylerjet, 2024-02-17:
> I had klipper backup working fine for a while and then it stopped pushing backups altogether. After I realized this I manually ran the script so I'd get the verbose output and found similar set of errors. The compression step takes about 30 minutes and then it fails to push. It's just trying to backup my config folder, but when I comment out that folder in the .env and just try to have it backup the printer.cfg single file I get the same error.
> 
> ![ScreenShot 2024-02-17 at 12 28 56 PM](https://private-user-images.githubusercontent.com/142825267/305666405-ab1a7aad-48b4-4870-b549-e4e2c6d249ff.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDgxOTg2NDksIm5iZiI6MTcwODE5ODM0OSwicGF0aCI6Ii8xNDI4MjUyNjcvMzA1NjY2NDA1LWFiMWE3YWFkLTQ4YjQtNDg3MC1iNTQ5LWU0ZTJjNmQyNDlmZi5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMjE3JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDIxN1QxOTMyMjlaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT05OGU5ZmZiZGM3OTU4ZGViNzgzNzI3Mzg2MTBiOTg3NjhiY2ZkMTAwNDk0NTllODRmMTg3MTM0NTYwZTA3ZDU3JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.ZjVDy0WlYDeas0e8oejKO568MoNBumb-Kn2WlpJCIsU)

If you go to your `$HOME/printer_data` folder and run `du -h` how large does it say your config folder is?

CallumRD1, 2024-02-17:
> If you go to your `$HOME/printer_data` folder and run `du -h` how large does it say your config folder is?
[image]

378 MB, but 222 of that is input shaper results that I don't really want backed up.

Tylerjet, 2024-02-17:
If you increase the postBuffer as Staubgeborener mentioned [above](https://github.com/Staubgeborener/klipper-backup/issues/51#issuecomment-1930079504) do you still get the error. it may be that because its still trying to push the deletion of those files to the remote its not able to complete.

Another option to try would be to go into config_backup and reset the repo to the latest commit `git reset --hard` then on github delete all the files there and go back to config_back and do a `git pull` and then try the script again.

CallumRD1, 2024-02-17:
> If you increase the postBuffer as Staubgeborener mentioned [above](https://github.com/Staubgeborener/klipper-backup/issues/51#issuecomment-1930079504) do you still get the error. it may be that [...]

Those large files don't even exist on the repo. It was never able to get them to upload properly in the first place. The last commit that was successful was before the shake tune folders were added.

[image]

Tylerjet, 2024-02-17:
so lets try this in the config_backup folder, do `git reset --hard origin/main` this should reset your local repo to match what is currently on the remote as screenshotted and then try the running script.

CallumRD1, 2024-02-17:
That appears to have worked. 

I had modified the .env so that it's now only backing up the specific files I care most about, and that change is now mirrored on git. It didn't try to backup the shake tune files.

Tylerjet, 2024-02-17:
NICE! yeah sounds like it was just the cached commits wanting to push all that data. Shouldn't be terribly hard to introduce excluding certain folders or just also skipping jpg's and csv's and .zips like we do for sym-linked files.

CallumRD1, 2024-02-17:
It was almost certainly the pile of ~30 MB .csv files. If there was a way to easily exclude those, I'd be very happy. Automatically backing up the ~1 MB .png input shaper results would be useful, but I definitely don't need the raw data.

Staubgeborener, 2024-02-17:
Read [this](https://staubgeborener.github.io/klipper-backup/configuration/#gitignore) to exclude specific files.

Tylerjet, 2024-02-17:
Ahh yeah totally spaced you can put the files into .gitignore lol

Tylerjet, 2024-02-17:
actually that wouldn't work in this scenario, as that would be modifying .gitignore in the klipper_backup folder making it show as dirty in moonraker

Staubgeborener, 2024-02-17:
Dann, never thought about this issue. We should fix this as `.gitignore` and the corresponding docs are somehow pointless in this case.

CallumRD1, 2024-02-17:
Is it possible to put exclusions in the .env file similarly to how we choose what to include?

Tylerjet, 2024-02-17:
I am working on that right now actually should be an easy change as we are already checking against a regex for the extra printer.cfg so just making that a variable that can instead be set in .env so you can then just add all the file types and file names you want ignored.

Tylerjet, 2024-02-18:
@CallumRD1 in the klipper-backup folder could you do `git pull origin pull/62/head` that will pull in the pr i just made that now has an exclude variable, and neatly enough does utilize .gitignore so win win.

Staubgeborener, 2024-02-25:
Now we need to adjust the docs.

Staubgeborener, 2024-02-25:
@CallumRD1 Please update to the [newest version](https://github.com/Staubgeborener/klipper-backup/pull/66), you can adjust the `.env` file to update the `.gitignore` in order to exclude more files flexible.

closed 2024-03-25 by github-actions[bot]