Staubgeborener/Klipper-Backup #53: [Feature request]: Execute automatically when config changed

Tomblarom, 2024-02-09:
### Is your feature request related to a problem? Please describe.

[Backups on boot or with Cron](https://staubgeborener.github.io/klipper-backup/automation/) are fine, but actually I just need a backup when I changed things in one of the config-files.

### Describe the solution you'd like

It would be ideal to trigger a backup, only when a config-file has been added, modified or deleted. However, many of my config-files are co-related, so during major changes there should be a marco/switch, to temporarily disable the "update-on-change-feature" .

### Describe alternatives you've considered

Great project! Looked forward to something like this for a looong time! 😊

Staubgeborener, 2024-02-09:
That's a pretty cool idea. Should be possible with the `inotify` subsystem and a systemd service. I'll definitely think about it and leave this feature request open.

Staubgeborener, 2024-02-14:
@Tylerjet has created a good solution with `inotify` and a systemd service file. See the new docs section [Backup on file changes](https://staubgeborener.github.io/klipper-backup/automation/#backup-on-file-changes).

closed 2024-02-14 by Staubgeborener

Tomblarom, 2024-02-25:
> @Tylerjet has created a good solution with `inotify` and a systemd service file.
WOW! Thanks @Tylerjet. Very clean solution. Will test soon :)