Staubgeborener/Klipper-Backup #60: [Bug]: Script asks for password

stephenmott, 2024-02-17:
### What happened

I'm sure I'm doing something dumb, but the script kept asking for a password (3 times) then gives "remote: Repository not found" followed by "fatal: Authentication failed for 'https://github.com/stephenmott/hictop-klipper2.git/'".

### What did you expect to happen

Expected the script to not ask for passwords

### How to reproduce

I'm not sure..

### Additional information

The repo does exist and the password I'm using is the access token and if I go into the config_backuo folder and do a "git remote remove origin", "git remote add origin ..." and "git remote set-url origin ..." with the correct URL, followed by "git push -u origin main" it works fine..?

Then as soon as I run the script again it removes the token from the origin url, then asks for a password again and fails..?

The really odd thing is I'm doing this on 4 Pi's, (3 x Pi 4B & 1 x Pi 5B), the first one worked fine, no problem at all, the second and third gave this issue, but after I cleared git credentials and started again, they eventually worked (nor really sure why), one with a correct origin url, then other one wrong, missing the token. The fourth one just refuses to work no matter what I try. 

Not sure if it makes any difference but the one that won't work is the Pi 5B.

Tylerjet, 2024-02-17:
did you set the token in the .env file inside of klipper-backup?

Staubgeborener, 2024-02-17:
> https://staubgeborener.github.io/klipper-backup/configuration/#gitignore

Shouldn't matter, the script should work on nearly every Unix based system no matter which architecture is in use.

Screenshots are very helpful for debugging purpose. You followed the [docs](https://staubgeborener.github.io/klipper-backup/) for setup?

stephenmott, 2024-02-17:
Yes I worked through the docs, it worked fine the first time, all I effectively did for the other 3 was change the repository name.. This is what it does on the third one:

[image]

Staubgeborener, 2024-02-17:
Ich you please post your `.env` file with hidden token. Did you add the .git extension to your repository name?

stephenmott, 2024-02-17:
This is the .env file:

github_topken=ghp_XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
github_username=stephenmott
github_repository=hictop-klipper2
branch_name=main
commit_username="stephenmott"
commit_email="XXX@XXXX.com"

# Indivdual file syntax:
#  Note: script.sh starts its search in $HOME which is /home/{username}/
# Using below example the script will search for: /home/{username}/printer_data/config/printer.cfg

#path_printercfg=printer_data/config/printer.cfg

# Backup folder syntax:
#  Note: script.sh starts its search in $HOME which is /home/{username}/
# Using below example the script will search for: /home/{username}/printer_data/config/*
# `/*` should always be at the end of the path when backing up a folder so that the files inside of the folder are properly filtered and searched

path_klipperdata=printer_data/config/*

Tylerjet, 2024-02-17:
github_topken is the issue there is a typo there should be github_token

stephenmott, 2024-02-17:
Oh ffs, well spotted..

Staubgeborener, 2024-02-17:
If everything works again, please close the issue.

stephenmott, 2024-02-17:
Thanks all, yep all working, I knew I was doing something dumb..

closed 2024-02-17 by stephenmott