Staubgeborener/Klipper-Backup #63: [Bug]:

mcfool12, 2024-02-18:
### What happened

Failure to launch Backup on file changes

### What did you expect to happen

It to backup on change but the service wouldn't start

### How to reproduce

Use the current code

### Additional information

This
inotifywait -mrP -e close_write -e move -e delete --exclude "$exclude_pattern" $watchlist | \
should be
inotifywait -mr -e close_write -e move -e delete --exclude "$exclude_pattern" $watchlist | \
in klipper-backup-filewatch.service

Tylerjet, 2024-02-18:
`inotifywait -mrP` is correct do you have screenshots or can copy and paste the error that you were getting? -P tells inotifywait not to watch symbolic linked files/folders.

mcfool12, 2024-02-18:
● klipper-backup-filewatch.service - Klipper Backup Filewatch Service
   Loaded: loaded (/etc/systemd/system/klipper-backup-filewatch.service; enabled; vendor p>
   Active: inactive (dead) since Sun 2024-02-18 11:00:02 EST; 21s ago
  Process: 1700 ExecStart=/bin/bash -c   watchlist="";   while IFS= read -r path; do>
' | sort -u | tr '
' ' ');   exclude_pattern=".swp|.tmp|printer-[0-9]*_[0-9]*.cfg|.bak|.bkp";   inotifywa>
  Main PID: 1700 (code=exited, status=0/SUCCESS)
    CPU: 64ms

Feb 18 11:00:02 voron systemd[1]: Started Klipper Backup Filewatch Service.
Feb 18 11:00:02 voron bash[1711]: inotifywait: invalid option -- 'P'
Feb 18 11:00:02 voron systemd[1]: klipper-backup-filewatch.service: Succeeded.

mcfool12, 2024-02-18:
Works fine without it.

Tylerjet, 2024-02-18:
if you do `Inotifiywait --help` does `--no-dereference` show up in the list of options?

Tylerjet, 2024-02-18:
Version 3.22.6.0 seems to list -P as a valid option
```
inotifywait 3.22.6.0
Wait for a particular event on a file or set of files.
Usage: inotifywait [ options ] file1 [ file2 ] [ file3 ] [ ... ]
Options:
        -h|--help       Show this help text.
        @<file>         Exclude the specified file from being watched.
        --exclude <pattern>
                        Exclude all events on files matching the
                        extended regular expression <pattern>.
                        Only the last --exclude option will be
                        taken into consideration.
        --excludei <pattern>
                        Like --exclude but case insensitive.
        --include <pattern>
                        Exclude all events on files except the ones
                        matching the extended regular expression
                        <pattern>.
        --includei <pattern>
                        Like --include but case insensitive.
        -m|--monitor    Keep listening for events forever or until --timeout expires.
                        Without this option, inotifywait will exit after one event is received.
        -d|--daemon     Same as --monitor, except run in the background
                        logging events to a file specified by --outfile.
                        Implies --syslog.
        -P|--no-dereference
                        Do not follow symlinks.
        -r|--recursive  Watch directories recursively.
        --fromfile <file>
                        Read files to watch from <file> or `-' for stdin.
        -o|--outfile <file>
                        Print events to <file> rather than stdout.
        -s|--syslog     Send errors to syslog rather than stderr.
        -q|--quiet      Print less (only print events).
        -qq             Print nothing (not even events).
        --format <fmt>  Print using a specified printf-like format
                        string; read the man page for more details.
        --no-newline    Don't print newline symbol after
                        --format string.
        --timefmt <fmt> strftime-compatible format string for use with
                        %T in --format string.
        -c|--csv        Print events in CSV format.
        -t|--timeout <seconds>
                        When listening for a single event, time out after
                        waiting for an event for <seconds> seconds.
                        If <seconds> is zero, inotifywait will never time out.
        -e|--event <event1> [ -e|--event <event2> ... ]
                Listen for specific event(s).  If omitted, all events are
                listened for.

Exit status:
        0  -  An event you asked to watch for was received.
        1  -  An event you did not ask to watch for was received
              (usually delete_self or unmount), or some error occurred.
        2  -  The --timeout option was given and no events occurred
              in the specified interval of time.

Events:
        access          file or directory contents were read
        modify          file or directory contents were written
        attrib          file or directory attributes changed
        close_write     file or directory closed, after being opened in
                        writable mode
        close_nowrite   file or directory closed, after being opened in
                        read-only mode
        close           file or directory closed, regardless of read/write mode
        open            file or directory opened
        moved_to        file or directory moved to watched directory
        moved_from      file or directory moved from watched directory
        move            file or directory moved to or from watched directory
        move_self               A watched file or directory was moved.
        create          file or directory created within watched directory
        delete          file or directory deleted within watched directory
        delete_self     file or directory was deleted
        unmount         file system containing file or directory unmounted
```

mcfool12, 2024-02-18:
Hmm, appears that I am on 3.14 and showing no update available.

Tylerjet, 2024-02-18:
hrmm what distro are you running i know both my test instance and printer are debian based (ubuntu and armbian), and even they are technically behind the actual latest of inotify which is 4.23.9.0.

mcfool12, 2024-02-18:
pi@voron:~ $ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
NAME="Raspbian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=raspbian
ID_LIKE=debian

Staubgeborener, 2024-02-18:
Maybe run

```shell
sudo apt update
sudo apt install inotify-tools
```

mcfool12, 2024-02-18:
Yea apt update is where it is showing no updates and install shows no updates "inotify-tools is already the newest version (3.14-8.1)".

mcfool12, 2024-02-23:
Sorry for the delay.  That got it, thanks.

Tylerjet, 2024-02-24:
Awesome I'll be sure to add that to the docks as well.

Tomblarom, 2024-02-25:
Unfortunately I can't get the service to start..

```
pi@rpi3-3d-ender:~ $ sudo systemctl status klipper-backup-filewatch.service
● klipper-backup-filewatch.service - Klipper Backup Filewatch Service
   Loaded: loaded (/etc/systemd/system/klipper-backup-filewatch.service; enabled; vendor preset: enabled)
   Active: inactive (dead)

Feb 25 14:43:52 rpi3-3d-ender systemd[1]: Started Klipper Backup Filewatch Service.
Feb 25 14:43:52 rpi3-3d-ender bash[743]: inotifywait: symbol lookup error: inotifywait: undefined symbol: inotifytools_init
```

Staubgeborener, 2024-02-25:
What version are you running? Run `inotifywait -h`. I think it's an old one, see [this post](https://github.com/Staubgeborener/klipper-backup/issues/63#issuecomment-1951540043). I will adjust the docs later.

Tomblarom, 2024-02-25:
The error occured with the latest `4.23.9.0`. I had `3.14` installed previously and I removed it by running `sudo apt-get purge inotify-tools`.  Then I downloaded the [release zip of 4.23.9.0](https://github.com/inotify-tools/inotify-tools/releases/tag/4.23.9.0) and did the install as described except I used the zip and this command `./configure --prefix=/usr --libdir=/lib64`. I don't what did the trick, but it works now :)

```
pi@rpi3-3d-ender:~/inotify-tools-4.23.9.0 $ journalctl -u klipper-backup-filewatch.service
-- Logs begin at Sun 2024-02-25 14:43:40 GMT, end at Sun 2024-02-25 15:09:51 GMT. --
Feb 25 14:43:52 rpi3-3d-ender systemd[1]: Started Klipper Backup Filewatch Service.
Feb 25 14:43:52 rpi3-3d-ender bash[743]: inotifywait: symbol lookup error: inotifywait: undefined symbol: inotifytools_init
Feb 25 14:43:52 rpi3-3d-ender systemd[1]: klipper-backup-filewatch.service: Succeeded.
Feb 25 14:49:56 rpi3-3d-ender systemd[1]: Started Klipper Backup Filewatch Service.
Feb 25 14:49:56 rpi3-3d-ender bash[5697]: inotifywait: symbol lookup error: inotifywait: undefined symbol: inotifytools_init
Feb 25 14:49:56 rpi3-3d-ender systemd[1]: klipper-backup-filewatch.service: Succeeded.
Feb 25 15:07:15 rpi3-3d-ender systemd[1]: Started Klipper Backup Filewatch Service.
Feb 25 15:07:15 rpi3-3d-ender bash[10136]: Setting up watches.  Beware: since -r was given, this may take a while!
Feb 25 15:07:15 rpi3-3d-ender bash[10136]: Watches established.
Feb 25 15:08:19 rpi3-3d-ender bash[10136]: Event Type: MOVED_TO, Watched Path: /home/pi/printer_data/config/, File Name: printer.cfg
```

Tomblarom, 2024-02-25:
Are you getting this `undefined symbol error` in the logs as well with your installation?

Staubgeborener, 2024-02-25:
> Are you getting this `undefined symbol error` in the logs as well with your installation?

No.

ChrisToxz, 2024-03-28:
After updating it manually to 4.23 I also get ` inotifywait: symbol lookup error: inotifywait: undefined symbol: inotifytools_init`.
And my service exited after starting. 
I deleted inotifywait again and installed version 3.22.60, the one that @Tylerjet mentioned using [here](https://github.com/Staubgeborener/klipper-backup/issues/63#issuecomment-1951403750). 
And with 3.22.6.0 it works all fine!

After cloning inotifywait I checked out `c8bdbc0`

So threeconclusion.

1. Raspbian 11 comes with 3.14. (Maybe newer version available in Raspbian 12)
2. Latest version of `inotifywait` 4.23.9 comes with an `undefined symbol error`
3. Version 3.22.6.0 works as a charm (CC: @Tomblarom)

Staubgeborener, 2024-03-28:
@ChrisToxz following [this recommended](https://github.com/Staubgeborener/klipper-backup/issues/63#issuecomment-1951540043) instructions will install inotify with the latest version (currently: 4.23.9.0). This should work and there should be no need to use version 3.22.6.0 - could you test this? Right now we recommend those steps in the docs.

ChrisToxz, 2024-03-28:
@Staubgeborener 
I did follow those instructions, and with 4.23.9.0 I got the 'undefined symbol' error.
I removed the installation, followed the same steps, but after cloning i did `checkout c8bdbc0`, which is the 3.22.6.0 release, after that everything worked without errors.

With both 3.14 and 4.23.9.0 I didn't get the file watcher working at all.
3.14 is clear, missing the `-P` param, 4.23.9.0, at least for me, the undefined symbol error.
I am still using Raspbian 11, which is a bit old, and might causing it.

Staubgeborener, 2024-03-28:
Thanks. We have to figure out about the origin of this issue and maybe rewrite the script (in worst case it's systems related). But to this point i will add the `git checkout c8bdbc0` line to the recommend instructions.

Tylerjet, 2024-03-28:
@ChrisToxz @Tomblarom Were you seeing any errors at the end of the manual build process for 4.23.9?

Tylerjet, 2024-03-28:
I would also make sure to completely remove the apt-get version if you have not already what may be happening is you did the manual install but its still referencing the apt-get version which then combines up to date and out of date packages which could give the error.

Tylerjet, 2024-03-28:
https://github.com/inotify-tools/inotify-tools/issues/162#issuecomment-1069393650

ChrisToxz, 2024-03-28:
@Tylerjet 
There are no errors during the build process at all.

`sudo systemctl status klipper-backup-filewatch.service`
```
klipper-backup-filewatch.service - Klipper Backup Filewatch Service
     Loaded: loaded (/etc/systemd/system/klipper-backup-filewatch.service; enabled; vendor preset: enabled)
     Active: inactive (dead) since Thu 2024-03-28 12:45:54 GMT; 4s ago
    Process: 1370 ExecStart=/bin/bash -c      watchlist="";      while IFS= read -r path; do          for file in $path; do              if [ ! -h "$file" ]; then                  file_dir=$(dirname "$file");                  if [ "$file_dir" = "." ]; then                      watchlist+=" $HOME/$f>
' | sort -u | tr '
' ' ');      exclude_pattern=".swp|.tmp|printer-[0-9]*_[0-9]*.cfg|.bak|.bkp";      inotifywait -mr -e close_write -e move -e delete --exclude "$exclude_pattern" $watchlist |      while read -r path event file; do          if [ -z $file ]; then              file=$(basename "$path");          fi;    >
   Main PID: 1370 (code=exited, status=0/SUCCESS)
        CPU: 58ms

Mar 28 12:45:54 raspberrypi systemd[1]: Started Klipper Backup Filewatch Service.
Mar 28 12:45:54 raspberrypi bash[1381]: inotifywait: symbol lookup error: inotifywait: undefined symbol: inotifytools_init
Mar 28 12:45:54 raspberrypi systemd[1]: klipper-backup-filewatch.service: Succeeded.
```

Thanks for the inotify issue.
The timestamp is indeed earlier in the day. Seems like purge didn't really did his job.
```
lrwxrwxrwx  1 root root     24 Mar 28 12:54 libinotifytools.so -> libinotifytools.so.0.4.1
lrwxrwxrwx  1 root root     24 Mar 28 12:54 libinotifytools.so.0 -> libinotifytools.so.0.4.1
-rwxr-xr-x  1 root root 120988 Mar 28 12:54 libinotifytools.so.0.4.1
```
 
If you guys get this all working with 4.23.9 out of the box, I would consider this indeed as an environmental issue.
When I got time I will clean it up, to see if it works with 4.23.9.

Thanks for the quick replies!

ChrisToxz, 2024-03-28:
Correction: The lib files are not from a previous installation, as https://github.com/inotify-tools/inotify-tools/issues/162#issuecomment-1069393650 might suggest as cause.

My first successful commit of the file watcher was at 12:56, and the libs has been installed at 12:54.

So when I installed 3.22.6.0, they got overwritten (or they are just new and purged did its job)

Tylerjet, 2024-03-28:
Hrmm yeah if you have the chance could you try clearing just anything i notify related and just running the following i notice that in my old above comment only ```make install``` had sudo but all of them may need to be run as sudo which could affect it (maybe?):

```
git clone https://github.com/inotify-tools/inotify-tools.git

sudo apt-get install autoconf autotools-dev automake libtool

cd inotify-tools/

sudo ./autogen.sh
sudo ./configure --prefix=/usr
sudo make
sudo make install

cd ..
sudo rm -rf inotify-tools
```

I do have a pi running debian 11 that does not have the apt-get version installed so i will try and check there later on and see if i might be able to get the error