Staubgeborener/Klipper-Backup #64: [Bug]:

TItan371, 2024-02-22:
### What happened

// ! [rejected] main -> main (non-fast-forward)
01:11:31 PM 
// error: failed to push some refs to 'https://github.com/TItan371/MinionKlipperBackUP.git'
01:11:31 PM 
// hint: Updates were rejected because the tip of your current branch is behind
// hint: its remote counterpart. Integrate the remote changes (e.g.
// hint: 'git pull ...') before pushing again.
// hint: See the 'Note about fast-forwards' in 'git push --help' for details.
01:11:31 PM 
// Command {update_git_script} finished

### What did you expect to happen

update finish

### How to reproduce

run script.sh

### Additional information

01:11:25 PM 
// Running Command {update_git_script}...:
01:11:26 PM 
// Klipper-backup is up to date
01:11:27 PM 
// hint: Pulling without specifying how to reconcile divergent branches is
// hint: discouraged. You can squelch this message by running one of the following
// hint: commands sometime before your next pull:
// hint:
// hint: git config pull.rebase false # merge (the default strategy)
// hint: git config pull.rebase true # rebase
// hint: git config pull.ff only # fast-forward only
// hint:
// hint: You can replace "git config" with "git config --global" to set a default
// hint: preference for all repositories. You can also pass --rebase, --no-rebase,
// hint: or --ff-only on the command line to override the configured default per
// hint: invocation.
01:11:29 PM 
// From https://github.com/TItan371/MinionKlipperBackUP
// * branch main -> FETCH_HEAD
01:11:29 PM 
// 1c88d65..9e10ff9 main -> origin/main
01:11:29 PM 
// fatal: refusing to merge unrelated histories
01:11:29 PM 
// Skipping symbolic link: printer_data/config/fluidd.cfg
01:11:29 PM 
// Skipping symbolic link: printer_data/config/KAMP
01:11:29 PM 
// Skipping symbolic link: printer_data/config/K-ShakeTune
01:11:29 PM 
// Skipping symbolic link: printer_data/config/timelapse.cfg
01:11:29 PM 
// On branch main
// nothing to commit, working tree clean
01:11:30 PM 
// From https://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@github.com/TItan371/MinionKlipperBackUP.git
01:11:31 PM 
// To https://github.com/TItan371/MinionKlipperBackUP.git
// ! [rejected] main -> main (non-fast-forward)
01:11:31 PM 
// error: failed to push some refs to 'https://github.com/TItan371/MinionKlipperBackUP.git'
01:11:31 PM 
// hint: Updates were rejected because the tip of your current branch is behind
// hint: its remote counterpart. Integrate the remote changes (e.g.
// hint: 'git pull ...') before pushing again.
// hint: See the 'Note about fast-forwards' in 'git push --help' for details.
01:11:31 PM 
// Command {update_git_script} finished

​env file 

github_username=TItan371
github_repository=MinionKlipperBackUP
branch_name=main
commit_username="TItan371"

Tylerjet, 2024-02-22:
You will definitely want to create a new token/ you may already have to, as github has likely revoked it since it was still in the remote link when you copied the script output. 

in the klipper-backup folder if you do git log what is the latest commit message you see?

TItan371, 2024-02-22:
:~/klipper-backup $  git log
commit 67929a3e868279547ead4d4f55f2be8ab352f241 (HEAD -> main, tag: v1.2.1, origin/main, origin/HEAD)
Author: Staubgeborener <eric.schroeder@protonmail.com>
Date:   Wed Feb 14 19:39:02 2024 +0100

    small changes in readme + beautify code

commit 6c15e9d96150413c648a302c5407b5d8c06b587a
Author: MrTylerjet <mrtylerjet@gmail.com>
Date:   Sat Feb 10 13:13:26 2024 -0700

    removed timezone check instead using %x to use locale's date representation, and %X for locales time representation (#56)

commit b2f7c17bc859a80406a5d71b02360ac06ff32bc1
Author: Florian Gareis <TheZoker@users.noreply.github.com>
Date:   Sat Feb 10 20:59:02 2024 +0100

    Add git url as parameter (#55)

commit 60cfea820cba334b9e0830455d85c2ac115800fe
Author: Staubgeborener <eric.schroeder@protonmail.com>
Date:   Sat Feb 10 16:47:57 2024 +0100

    check if ipconfig or ifconfig is installed

    stop unix bullying

commit 2346e4f996d9509b1c1ec0632499d4537d6cdf57
Author: MrTylerjet <mrtylerjet@gmail.com>
Date:   Wed Feb 7 04:15:57 2024 -0700

    instead of using git hashes for a unique id at the end of the email, use the mac address of the first network device to create a unique has instead that will always ensure a unique id (#52)

commit cb50055e3fee7cfb22d4ae235c72c00a5d93fd75 (tag: 1.2.0)
Merge: 519aa33 c82d3be
Author: Staubgeborener <eric.schroeder@protonmail.com>
Date:   Mon Jan 29 17:17:49 2024 +0100

    Merge branch 'main' of https://github.com/Staubgeborener/klipper-backup

commit 519aa339fda868f83025d82895b70ae7bbdc76c9
Author: Staubgeborener <eric.schroeder@protonmail.com>
Date:   Mon Jan 29 17:17:42 2024 +0100

    Redirect to new docs + update Video

    Again, big thanks to Minimal 3DP for the great YouTube Video!

:

TItan371, 2024-02-22:
[git.txt](https://github.com/Staubgeborener/klipper-backup/files/14369441/git.txt)

Tylerjet, 2024-02-22:
Could you try removing the config_backup folder and then try running the script again.

TItan371, 2024-02-22:
[git.txt](https://github.com/Staubgeborener/klipper-backup/files/14369865/git.txt)

Tylerjet, 2024-02-22:
Sorry meant running script.sh again. I only needed the first entry of got log just wanted to be sure it true was reporting it was up to date properly.

TItan371, 2024-02-22:
02:30:33 PM 
$ BACKUP_CFG
02:30:34 PM 
// Running Command {update_git_script}...:
02:30:35 PM 
// Klipper-backup is up to date
02:30:36 PM 
// hint: Pulling without specifying how to reconcile divergent branches is
// hint: discouraged. You can squelch this message by running one of the following
// hint: commands sometime before your next pull:
// hint:
// hint: git config pull.rebase false # merge (the default strategy)
// hint: git config pull.rebase true # rebase
// hint: git config pull.ff only # fast-forward only
// hint:
// hint: You can replace "git config" with "git config --global" to set a default
// hint: preference for all repositories. You can also pass --rebase, --no-rebase,
// hint: or --ff-only on the command line to override the configured default per
// hint: invocation.
02:30:37 PM 
// From https://github.com/TItan371/MinionKlipperBackUP
// * branch main -> FETCH_HEAD
02:30:37 PM 
// fatal: refusing to merge unrelated histories
02:30:37 PM 
// Skipping symbolic link: printer_data/config/fluidd.cfg
// Skipping symbolic link: printer_data/config/KAMP
02:30:37 PM 
// Skipping symbolic link: printer_data/config/K-ShakeTune
02:30:37 PM 
// Skipping symbolic link: printer_data/config/timelapse.cfg
02:30:37 PM 
// On branch main
// nothing to commit, working tree clean
02:30:39 PM 
// From https://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@github.com/TItan371/MinionKlipperBackUP.git
02:30:40 PM 
// To https://github.com/TItan371/MinionKlipperBackUP.git
// ! [rejected] main -> main (non-fast-forward)
02:30:40 PM 
// error: failed to push some refs to 'https://github.com/TItan371/MinionKlipperBackUP.git'
02:30:40 PM 
// hint: Updates were rejected because the tip of your current branch is behind
// hint: its remote counterpart. Integrate the remote changes (e.g.
// hint: 'git pull ...') before pushing again.
// hint: See the 'Note about fast-forwards' in 'git push --help' for details.
02:30:40 PM 
// Command {update_git_script} finished

TItan371, 2024-02-22:
02:30:33 PM 
$ BACKUP_CFG
02:30:34 PM 
// Running Command {update_git_script}...:
02:30:35 PM 
// Klipper-backup is up to date
02:30:36 PM 
// hint: Pulling without specifying how to reconcile divergent branches is
// hint: discouraged. You can squelch this message by running one of the following
// hint: commands sometime before your next pull:
// hint:
// hint: git config pull.rebase false # merge (the default strategy)
// hint: git config pull.rebase true # rebase
// hint: git config pull.ff only # fast-forward only
// hint:
// hint: You can replace "git config" with "git config --global" to set a default
// hint: preference for all repositories. You can also pass --rebase, --no-rebase,
// hint: or --ff-only on the command line to override the configured default per
// hint: invocation.
02:30:37 PM 
// From https://github.com/TItan371/MinionKlipperBackUP
// * branch main -> FETCH_HEAD
02:30:37 PM 
// fatal: refusing to merge unrelated histories
02:30:37 PM 
// Skipping symbolic link: printer_data/config/fluidd.cfg
// Skipping symbolic link: printer_data/config/KAMP
02:30:37 PM 
// Skipping symbolic link: printer_data/config/K-ShakeTune
02:30:37 PM 
// Skipping symbolic link: printer_data/config/timelapse.cfg
02:30:37 PM 
// On branch main
// nothing to commit, working tree clean
02:30:39 PM 
// From https://github@github.com/TItan371/MinionKlipperBackUP.git
02:30:40 PM 
// To https://github.com/TItan371/MinionKlipperBackUP.git
// ! [rejected] main -> main (non-fast-forward)
02:30:40 PM 
// error: failed to push some refs to 'https://github.com/TItan371/MinionKlipperBackUP.git'
02:30:40 PM 
// hint: Updates were rejected because the tip of your current branch is behind
// hint: its remote counterpart. Integrate the remote changes (e.g.
// hint: 'git pull ...') before pushing again.
// hint: See the 'Note about fast-forwards' in 'git push --help' for details.
02:30:40 PM 
// Command {update_git_script} finished

TItan371, 2024-02-22:
or this?

$ ~/klipper-backup/script.sh
Klipper-backup is up to date

Password for 'https://github_pat_@github.com':
Skipping symbolic link: printer_data/config/fluidd.cfg
Skipping symbolic link: printer_data/config/KAMP
Skipping symbolic link: printer_data/config/K-ShakeTune
Skipping symbolic link: printer_data/config/timelapse.cfg
On branch main
nothing to commit, working tree clean
Password for 'https://github_pat_@github.com':
remote: Invalid username or password.
fatal: Authentication failed for 'https://github.com/TItan371/MinionKlipperBackUP.git/'
Password for 'https://github_pat_@github.com':
remote: Invalid username or password.
fatal: Authentication failed for 'https://github.com/TItan371/MinionKlipperBackUP.git/'
pi@MinionPi:~ $ ~/klipper-backup/script.sh
Klipper-backup is up to date

Password for 'https://github_pat_@github.com':
Skipping symbolic link: printer_data/config/fluidd.cfg
Skipping symbolic link: printer_data/config/KAMP
Skipping symbolic link: printer_data/config/K-ShakeTune
Skipping symbolic link: printer_data/config/timelapse.cfg
On branch main
nothing to commit, working tree clean
Password for 'https://github_pat_@github.com':
remote: Support for password authentication was removed on August 13, 2021.
remote: Please see https://docs.github.com/en/get-started/getting-started-with-git/about-remote-repositories#cloning-with-https-urls for information on currently recommended modes of authentication.
fatal: Authentication failed for 'https://github.com/TItan371/MinionKlipperBackUP.git/'
Password for 'https://github_pat_@github.com':
remote: Support for password authentication was removed on August 13, 2021.
remote: Please see https://docs.github.com/en/get-started/getting-started-with-git/about-remote-repositories#cloning-with-https-urls for information on currently recommended modes of authentication.
fatal: Authentication failed for 'https://github.com/TItan371/MinionKlipperBackUP.git/'

TItan371, 2024-02-22:
03:25:56 PM 
$ BACKUP_CFG
03:25:56 PM 
// Running Command {update_git_script}...:
03:25:57 PM 
// Klipper-backup is up to date
03:25:58 PM 
// hint: Pulling without specifying how to reconcile divergent branches is
// hint: discouraged. You can squelch this message by running one of the following
// hint: commands sometime before your next pull:
// hint:
// hint: git config pull.rebase false # merge (the default strategy)
// hint: git config pull.rebase true # rebase
// hint: git config pull.ff only # fast-forward only
// hint:
// hint: You can replace "git config" with "git config --global" to set a default
// hint: preference for all repositories. You can also pass --rebase, --no-rebase,
// hint: or --ff-only on the command line to override the configured default per
// hint: invocation.
03:25:59 PM 
// From https://github.com/TItan371/MinionKlipperBackUP
// * branch main -> FETCH_HEAD
03:26:00 PM 
// fatal: refusing to merge unrelated histories
03:26:00 PM 
// Skipping symbolic link: printer_data/config/fluidd.cfg
// Skipping symbolic link: printer_data/config/KAMP
03:26:00 PM 
// Skipping symbolic link: printer_data/config/K-ShakeTune
03:26:00 PM 
// Skipping symbolic link: printer_data/config/timelapse.cfg
03:26:00 PM 
// On branch main
// nothing to commit, working tree clean
03:26:01 PM 
// From https://github_pat_@github.com/TItan371/MinionKlipperBackUP.git
03:26:02 PM 
// To https://github.com/TItan371/MinionKlipperBackUP.git
// ! [rejected] main -> main (non-fast-forward)
03:26:02 PM 
// error: failed to push some refs to 'https://github.com/TItan371/MinionKlipperBackUP.git'
03:26:02 PM 
// hint: Updates were rejected because the tip of your current branch is behind
// hint: its remote counterpart. Integrate the remote changes (e.g.
// hint: 'git pull ...') before pushing again.
// hint: See the 'Note about fast-forwards' in 'git push --help' for details.
03:26:02 PM 
// Command {update_git_script} finished

Tylerjet, 2024-02-22:
where you previously running the klipper-backup script to that repository or was there already previous data and commits from another system/manual backups. Usually if you are getting a warning about histories not matching its because the commits vary to greatly and the histories are not aligned.

As well the password error is from not having the token set in .env

TItan371, 2024-02-22:
03:43:16 PM 
$ BACKUP_CFG
03:43:16 PM 
// Running Command {update_git_script}...:
03:43:17 PM 
// Klipper-backup is up to date
03:43:18 PM 
// hint: Pulling without specifying how to reconcile divergent branches is
// hint: discouraged. You can squelch this message by running one of the following
// hint: commands sometime before your next pull:
// hint:
// hint: git config pull.rebase false # merge (the default strategy)
// hint: git config pull.rebase true # rebase
// hint: git config pull.ff only # fast-forward only
// hint:
// hint: You can replace "git config" with "git config --global" to set a default
// hint: preference for all repositories. You can also pass --rebase, --no-rebase,
// hint: or --ff-only on the command line to override the configured default per
// hint: invocation.
03:43:19 PM 
// From https://github.com/TItan371/MinionKlipperBackUP
// * branch main -> FETCH_HEAD
03:43:19 PM 
// fatal: refusing to merge unrelated histories
03:43:19 PM 
// Skipping symbolic link: printer_data/config/fluidd.cfg
// Skipping symbolic link: printer_data/config/KAMP
03:43:19 PM 
// Skipping symbolic link: printer_data/config/K-ShakeTune
03:43:19 PM 
// Skipping symbolic link: printer_data/config/timelapse.cfg
03:43:19 PM 
// On branch main
// nothing to commit, working tree clean
03:43:21 PM 
// From https://github_pat_@github.com/TItan371/MinionKlipperBackUP.git
03:43:22 PM 
// To https://github.com/TItan371/MinionKlipperBackUP.git
// ! [rejected] main -> main (non-fast-forward)
// error: failed to push some refs to 'https://github.com/TItan371/MinionKlipperBackUP.git'
03:43:22 PM 
// hint: Updates were rejected because the tip of your current branch is behind
// hint: its remote counterpart. Integrate the remote changes (e.g.
// hint: 'git pull ...') before pushing again.
// hint: See the 'Note about fast-forwards' in 'git push --help' for details.
03:43:22 PM 
// Command {update_git_script} finished

TItan371, 2024-02-22:
i find this https://github.com/Staubgeborener/klipper-backup/issues/18.
can you help me where shold i put this cmd
 git push --force

Tylerjet, 2024-02-22:
> where you previously running the klipper-backup script to that repository or was there already previous data and commits from another system/manual backups. Usually if you are getting a warning [...]

As noted in the above were you using a different backup system before or was the repo you are trying to push to used for something else prior to using klipper-backup?

TItan371, 2024-02-22:
before this i use ssh to download all file and this is my first time use my github repositories.

Tylerjet, 2024-02-22:
Ahh ok so the best option I would say is to create a new repository to use the script with and then verify all the files you backed up match the old repository.

Then you can delete the old repository since it wouldn't be needed.

TItan371, 2024-03-08:
because i still can't get it work so i giveup , back to old backup way .

Staubgeborener, 2024-03-08:
You could try to delete the folders 

```shell
rm -rf ~/config_backup
rm -rf ~/klipper-backup
```

And follow [this](https://m.youtube.com/watch?v=47qV9BE2n_Y) pretty good video together with the docs.

closed 2024-04-06 by github-actions[bot]

Staubgeborener, 2024-08-12:
I guess eother one of these two things:

1. you didn't follow the [documentation](https://klipperbackup.xyz/)
2. or you try to push the files to a repository which already exists 

The error message is about having a repository at GitHub which has a different commit history as the local one. Mostly this happens while trying to push a commit to a repository which already exists with files.

So I would suggest the following: try `./script.sh --fix` and see if you can run the script afterwards without the `--fix` parameter.

If this doesn't work, try to push your backup to another repository. So create a new one on GitHub with a new token, rerun the installation script with the new information and push your backup to the new repository. If this works, delete the old repository because you will not need this anymore.