Staubgeborener/Klipper-Backup #65: Custom README.md

Ibot-11, 2024-02-24:
How can I add a custom readme? The current script always creates a new readme which overrides any changes.

Tylerjet, 2024-02-24:
at the moment you can't but it likely wouldn't be to hard to implement, how are you editing it right now via the browser? 

the likely implementation would be creating the README file in the backup folder(as it does currently) but then not deleting it like we do for the .git folder, that way changes are preserved.

As currently all files are deleted a the end of the script execution so that if a file is deleted before the next backup that deletion is pushed to the remote.  So we just recreate the README from scratch each time.

Staubgeborener, 2024-02-24:
Like @Tylerjet mentioned, it's not possible right now. Maybe i will support this in future with an optional `.env` parameter.

Staubgeborener, 2024-02-25:
@Ibot-11 Please update to the [newest version](https://github.com/Staubgeborener/klipper-backup/pull/66). You can edit the `README.md` file in your `~/config_backup` folder to create your own (you have to run the script at least once to create the initially `README.md`).

Ibot-11, 2024-02-25:
Thank you for the quick implementation!
Works like a charm.