Staubgeborener/Klipper-Backup #67: Skript asks for password after update to newest version

LoSt96, 2024-02-25:
### Code of Conduct

- [X] I promise that i have used markdown syntax to make the `code` more readable which helps to fix the problem

### What happened

I made a successful backup and the script showed, that a new version is available.

I updated with the command: cd ~/klipper-backup && git pull

After this update i get asked for a password when i want to do a backup.

### What did you expect to happen

Successful backup

### How to reproduce

Update from version 1.2.0 with the command: cd ~/klipper-backup && git pull

Staubgeborener, 2024-02-25:
Does the script ask for the sudo/root password or for the github password? Is it related to [this](https://github.com/Staubgeborener/klipper-backup/issues/60) closed issue? Please post a screenshot of this behavior and also your `.env` file with a hidden token.

LoSt96, 2024-02-25:
The script asks for the github password.
This is my .env file:
```
github_token=ghp_XXX
github_username=LoSt96
github_repository=Neptune_4_Plus
branch_name=main
commit_username="LoSt96"
commit_email="XXX@XXX"

# Indivdual file syntax:
#  Note: script.sh starts its search in $HOME which is /home/{username}/
# Using below example the script will search for: /home/{username}/printer_data/config/printer.cfg

#path_printercfg=printer_data/config/printer.cfg

# Backup folder syntax:
#  Note: script.sh starts its search in $HOME which is /home/{username}/
# Using below example the script will search for: /home/{username}/printer_data/config/*
# `/*` should always be at the end of the path when backing up a folder so that the files inside of the folder are prope$

path_klipperdata=klipper_config/*

# Array of strings in .gitignore pattern git format https://git-scm.com/docs/gitignore#_pattern_format for files that sh$
# New additions must be enclosed in double quotes and should follow the pattern format as noted in the above link
exclude=( \
"*.swp" \
"*.tmp" \
"printer-[0-9]*_[0-9]*.cfg" \
"*.bak" \
"*.bkp" \
"*.csv" \
"*.zip" \
)
```

[image]

I tried to use my github password when the script asks but then i get following error message: 

```
remote: Support for password authentication was removed on August 13, 2021.
remote: Please see https://docs.github.com/en/get-started/getting-started-with-git/about-remote-repositories#cloning-wi                    th-https-urls for information on currently recommended modes of authentication.
fatal: Authentication failed for 'https://ghp_XXX@github.com/LoSt96/Neptune_4_Plus.git/'
```

Staubgeborener, 2024-02-25:
Try to delete the .git config folder `sudo rm -r ~/config_backup/.git/config` (make a backup if this file beforehand and maybe also post the content here if this doesn't work) and try to run the script again.

LoSt96, 2024-02-25:
I removed the file, but it made no difference. 
The file is now containing this: 
```
[init]
    defaultBranch = main
[core]
        repositoryformatversion = 0
        filemode = true
        bare = false
        logallrefupdates = true
[user]
        name = LoSt96
        email = XXX@XXX.de
[remote "origin"]
        url = https://ghp_XXX@github.com/LoSt96/Neptune_4_Plus.git
        fetch = +refs/heads/*:refs/remotes/origin/*
[advice]
        skippedCherryPicks = false
[branch "main"]
        remote = origin
        merge = refs/heads/main
```

Staubgeborener, 2024-02-25:
tbh honest i would start from scratch before digging around:
```shell
cd ~
cp ~/klipper-backup/.env .
sudo rm -r ~/config_backup ~/klipper-backup
git clone https://github.com/Staubgeborener/klipper-backup/
cd ~/klipper-backup
cp ../.env .
chmod +x ./script.sh
./script.sh
```

LoSt96, 2024-03-09:
I started from scratch which did not help.
I uninstalled and installed git which did not help.
I created a new fine-grained github token which did not help.
Then i created a new github token, but this time the classic version. 
This solved my problem and everything is working now again.

Staubgeborener, 2024-03-09:
In this case you have chosen the wrong permissions for the fine-grained token, but glad you figured it out.

closed 2024-03-09 by Staubgeborener