Staubgeborener/Klipper-Backup #69: Backing up symbolic links

joshmikow, 2024-02-27:
I got everything installed tonight and it went well. I'm liking having an easy backup after a scare this weekend when I thought I lost all of my configs I've been tweaking for the past month. 

I noticed in the logs that it is skipping symbolic links which results in my printer_data/config/KAMP folder, mainsail.cfg and timelapse.cfg not being backed up. Is there a way to include these in the .env so they'll be backed up properly? I included the printer_data/config/KAMP folder specifically in the .env but it still skips the folder because it's a symbolic link.

My moonraker also shows the "Dirty" warning but it sounds like this should have been resolved. Not sure what to check to see if I missed a step on this part.

Thank you for the hard work put in to make this so easy for those of us who aren't sure where to start with backing up klipper easily.

Tylerjet, 2024-02-27:
Symbolic links are not backed up as they are read only and cannot be modified, files like mainsail and fluidd.cfg even note in their configs that changes should not be saved there as they are overwritten on updates

Tylerjet, 2024-02-27:
As for the dirty warning if you go to the klipper-backup folder and run ```git status``` it will tell you if there are any files that have been modified locally which is likely causing the dirty warning in moonraker

joshmikow, 2024-02-28:
Thank you for the explanation on the symbolic links. I didn't realize that was how they worked. I also wasn't aware of what .cfg gets rebuilt on install. Makes sense to me.

I also used the 'git status' and found that i had edited the .gitignore before I realized using the multiple lines in the .env for folders to backup. I restored it to the default and it looks to be good now.

Thank you for the quick response and explanations!

closed 2024-02-28 by joshmikow

Tylerjet, 2024-02-28:
No Problem!