Staubgeborener/Klipper-Backup #71: Error when Uploading - fatal: not a git repository

freak12321, 2024-02-27:
### Code of Conduct

- [X] I promise that i have used markdown syntax to make the `code` more readable which helps to fix the problem

### What happened

Not the best with GitHub - sorry.
I installed it following you video and got the following errormessage when executing the script.sh
`Klipper-backup is up to date

fatal: not a git repository (or any of the parent directories): .git
fatal: not a git repository (or any of the parent directories): .git
Branch: main in .env does not match the currently checked out branch of: .
fatal: not a git repository (or any of the parent directories): .git
fatal: not a git repository (or any of the parent directories): .git
fatal: not in a git directory
fatal: not in a git directory
fatal: not a git repository (or any of the parent directories): .git
fatal: not a git repository (or any of the parent directories): .git
fatal: not a git repository (or any of the parent directories): .git
Skipping symbolic link: printer_data/config/KAMP
Skipping symbolic link: printer_data/config/mainsail.cfg
Skipping file: printer_data/config/printer-20240102_101142.cfg
Skipping file: printer_data/config/printer-20240112_131407.cfg
Skipping file: printer_data/config/printer-20240226_085630.cfg
Skipping file: printer_data/config/printer-20240226_154211.cfg
Skipping symbolic link: printer_data/config/timelapse.cfg
fatal: not a git repository (or any of the parent directories): .git
fatal: not a git repository (or any of the parent directories): .git
fatal: not a git repository (or any of the parent directories): .git
fatal: No remote configured to list refs from.
fatal: not a git repository (or any of the parent directories): .git
fatal: not a git repository (or any of the parent directories): .git
`

### What did you expect to happen

Upload to GitHub

### How to reproduce

Don't Know :-(

Tylerjet, 2024-02-27:
Did you edit the .env file to add your GitHub token and the repository name?

It would be best to read through the docs as well and follow those steps
https://staubgeborener.github.io/klipper-backup/

freak12321, 2024-02-27:
Yes it did that, edited Username, Token and Repo.

Staubgeborener, 2024-02-27:
1. use markdown. Do not just check the "I use markdown" checkbox but don't use it
2. this is not an official video. It's not from me.

Did you create the repository in your github profile? If yes, please show the link to this repository. Next: post your env file (with hidden token).

nleco, 2024-03-06:
I am having the same issue.

I have a raspberry pi 4.  I ran the command here:
https://staubgeborener.github.io/klipper-backup/installation/#install-klipper-backup

I then created a token on github and changed my username and the repo I made.  I followed this video as well:
https://www.youtube.com/watch?v=47qV9BE2n_Y&ab_channel=ModBot

I get the same issue as reported above.

my repo is created as private, but I made it public for now: https://github.com/nleco/klipper-backup-i3
the commit username/email were filled out automatically it seems.

here is my env file.:

`github_token=ghp_RANDOMSTRINGSANDWHATNOT
github_username=nleco
github_repository=klipper-backup-i3
branch_name=main
commit_username="ssanchez"
commit_email="ssanchez@klipper-pi-sqj9y8h"

# Indivdual file syntax:
#  Note: script.sh starts its search in $HOME which is /home/{username}/
# Using below example the script will search for: /home/{username}/printer_data/config/printer.cfg

#path_printercfg=printer_data/config/printer.cfg

# Backup folder syntax:
#  Note: script.sh starts its search in $HOME which is /home/{username}/
# Using below example the script will search for: /home/{username}/printer_data/config/*
# `/*` should always be at the end of the path when backing up a folder so that the files inside of the folder are properly filtered and searched

path_klipperdata=printer_data/config/*

# Array of strings in .gitignore pattern git format https://git-scm.com/docs/gitignore#_pattern_format for files that should not be uploaded to the remote repo
# New additions must be enclosed in double quotes and should follow the pattern format as noted in the above link
exclude=( \
"*.swp" \
"*.tmp" \
"printer-[0-9]*_[0-9]*.cfg" \
"*.bak" \
"*.bkp" \
"*.csv" \
"*.zip" \
)`

Tylerjet, 2024-03-06:
> I am having the same issue.
>
> I have a raspberry pi 4. I ran the command here: https://staubgeborener.github.io/klipper-backup/installation/#install-klipper-backup
>
> I then created a token on github [...]

when you go to `cd ~` and do ls is there a folder name config_backup listed?

Tylerjet, 2024-03-06:
@nleco furthermore to that is there a .git folder inside of config_backup you will need to do `ls -la` withing the config_backup folder.

The error from above would mean that the script is going into config_backup and seeing a .git folder but the folder itself is not an initialized git directory. This would only be occuring if the script was halted on first run before it could run git init or if .git was added manually before running the backup script.

nleco, 2024-03-06:
Yes. I do see a folder "config_backup".  
`drwxr-xr-x  3 ssanchez ssanchez 4096 Mar  6 01:27 config_backup`

within the folder, there is a '.git' folder.

running 'git -v' shows unknown option -v.  
However, "git --version" shows "git version 2.30.2"

I am pretty knowledgable with git and programming. I didn't want to dig too deep into the code.  I'm going to try and remove the config backup dir and try again.

nleco, 2024-03-06:
I renamed the "config_backup" directory and ran the setup.sh script again. that resolved my issue.

Tylerjet, 2024-03-06:
Interesting, in the old (now renamed) config_backup folder what was the contents of .git interested so see if maybe I need to add an additional check or change up some code maybe.

nleco, 2024-03-07:
Here you go. I did install the klipper instance with the klipper raspberry pi imager sofware. maybe it already had it.
[config_back.zip](https://github.com/Staubgeborener/klipper-backup/files/14519344/config_back.zip)

pgregg88, 2024-03-07:
I had the same issue with an RPI 4 following all directions posted here.
I fixed it by adding quotes around all values in the .env file.

Tylerjet, 2024-03-07:
> I had the same issue with an RPI 4 following all directions posted here. I fixed it by adding quotes around all values in the .env file.

Was it deleting the folder or adding the quotes that fixed it?

as in theory, the quotes shouldn't matter as we encapsulate all the variables in quotes throughout script.sh when they are being used.

If you try starting fresh (deleting both config_backup and klipper_backup folders) and following steps 1-7 from here: https://staubgeborener.github.io/klipper-backup/quickstart/#installation

can you see if you still have to wrap quotes around the values in the .env

stefanpieter, 2024-03-17:
> I renamed the "config_backup" directory and ran the setup.sh script again. that resolved my issue.

This fixed the issue for me.
Thank you.

closed 2024-04-15 by github-actions[bot]