Staubgeborener/Klipper-Backup #75: Primary setup command doesn't work

charminULTRA, 2024-03-02:
### Code of Conduct

- [X] I promise that i have used markdown syntax to make the `code` more readable which helps to fix the problem

### What happened

I entered "git clone https://github.com/Staubgeborener/klipper-backup.git && chmod +x ./klipper-backup/script.sh && cp ./klipper-backup/.env.example ./klipper-backup/.env" into my terminal

### What did you expect to happen

I expected it would create the .env file in my folder, it only placed the ReadMe and the Script into my folder, and there was no .env file created.

### How to reproduce

SSH into Raspberry Pi, run the command, check the contents of the folder.

Tylerjet, 2024-03-02:
if you do `ls -la` within klipper-backup folder what files show up? .env is considered a hidden file so you will not see it with normal ls.

Tylerjet, 2024-03-02:
[image]

charminULTRA, 2024-03-02:
> if you do `ls -la` within klipper-backup folder what files show up? .env is considered a hidden file so you will not see it with normal ls.

I can see it in that list after running that command. Would be nice to just have a bit of clarity on what the setup command is doing, and where it's creating the folder, because I'm not super familiar with bash.

Staubgeborener, 2024-03-02:
`git clone https://github.com/Staubgeborener/klipper-backup.git && chmod +x ./klipper-backup/script.sh && cp ./klipper-backup/.env.example ./klipper-backup/.env`

This code clones this repository on your system in your current folder. After that it will make the `script.sh` file executable. After that it creates a new file `.env` with the content of `.env.example`. 

Files which begins with a dot `.` are called hidden files because, well, they are hidden. Just doing a `ls -l` will not show them, you have to use `ls -la` (so an additional `-a`) to see them.

charminULTRA, 2024-03-02:
> `git clone https://github.com/Staubgeborener/klipper-backup.git && chmod +x ./klipper-backup/script.sh && cp ./klipper-backup/.env.example ./klipper-backup/.env`
> 
> This code clones this repository on your system in your current folder. After that it will make the `script.sh` file executable. After that it creates a new file `.env` with the content of `.env.example`.
> 
> You could just do a `cd ~` before running the code above just to be sure you are in your home directory.
> 
> Files which begins with a dot `.` are called hidden files because, well, they are hidden. Just doing a `ls -l` will not show them, you have to use `ls -la` (so an additional `-a`) to see them.
> 
> There is also a small section about this in the [docs](https://staubgeborener.github.io/klipper-backup/configuration/#how-can-i-edit-files-in-terminal).

Oh OK, I didn't execute this in the home directory, I did it in ~/klipper/, which seems to have messed things up when I try to do a manual backup.

charminULTRA, 2024-03-02:
When I try to run the command in the home directory, it says the folder already exists even though I can't find it.

Staubgeborener, 2024-03-02:
You just said, you run this command in `~/klipper-backup/` which messed up thing. This means you have a `klipper-backup` folder in your home directory.

```shell
cd ~
ls -l
```

I would recommend to delete this folder (which should exist) `sudo rm -r ~/klipper-backup` and beginn from scratch.

Staubgeborener, 2024-03-02:
I would recommend to start from scratch. I don't know in which folder you are right now and which subfolders are located there.

```shell
cd ~
ls -l #check if you see the Klipper-Backup folder or not
sudo rm -r klipper-backup #delete folder
```
And do this while `git clone` stuff again.
If you get any error messages please provide screenshots.

charminULTRA, 2024-03-02:
@Staubgeborener Thanks I really appreciate the personal help. I didn't realize that klipper-backup is somehow hidden in WinSCP (first screen is terminal, 2nd is WinSCP). I'll try this from scratch, thank you.

[image]

[image]

Tylerjet, 2024-03-02:
winscp is a bit of a pain sometimes where you have to refresh the remote list to see new folders

Staubgeborener, 2024-03-02:
Yeah, please just use your shell right now for this.
If you need any help, feel free to ask. Otherwise please close this issue.

charminULTRA, 2024-03-02:
Got it all working, thanks @Staubgeborener. One other pitfall I found was that checking create a Readme when creating a new Repo on Github web will also break the process. All set thanks.

closed 2024-03-02 by charminULTRA

Staubgeborener, 2024-03-02:
> One other pitfall I found was that checking create a Readme when creating a new Repo on Github web will also break the process

Stuff like this is important for us to know so we can adjust the documentation accordingly.