Staubgeborener/Klipper-Backup #77: fatal: loose object, and related errors

Bradford1040, 2024-03-02:
### Code of Conduct

- [X] I promise that i have used markdown syntax to make the `code` more readable which helps to fix the problem

### What happened

I had the backup set up on my main printer which I've used KIAUH on MX Linux (using SystemD kernel) it was working last month and now no longer works, I tried reinstalling on secondary printer and get the same error, following the install instructions to the letter, to make sure I had not done something wrong, I get the same error on the secondary system as I did the first, I have tried a few google search result fixes and nothing seems to fix the issue

### What did you expect to happen

for the macro to backup my my config's to github

### How to reproduce

just running it via macro or even running it via terminal <script.sh>

Staubgeborener, 2024-03-02:
Please provide your `.env` file with hidden token and post some screenshots.

Bradford1040, 2024-03-02:
the hidden token removed or present? I mean I guess I can just remake a new one after I post the one I am using now?

Bradford1040, 2024-03-02:
[image]
[copy_of_env.md](https://github.com/Staubgeborener/klipper-backup/files/14470939/copy_of_env.md)

Bradford1040, 2024-03-02:
I used terminal to run the backup instead of using macro inside mainsail, thought it would have more info, guess it doesn't

Bradford1040, 2024-03-02:
[image]
this is the second printer with fresh install and has the same error, there is NO folder 7E or file

Staubgeborener, 2024-03-02:
Just the `.env` and do some xxx where the token is placed.

> warning: adding embedded git repository: printer_ 1_data/config/klipper-macros

Interesting.

Bradford1040, 2024-03-02:
fatal: loose object 7e47384e1121e1110509f34bfd6e17b382398a3d (stored in .git/objects/7e/47384e1121e1110509f34bfd6e17b382398a3d) is corrupt

I have no clue where this file or folder is, or what it is?

Bradford1040, 2024-03-02:
> Just the `.env` and do some xxx where the token is placed.
> 
> > warning: adding embedded git repository: printer_ 1_data/config/klipper-macros
> 
> Interesting.

I jumped the gun lol, so will be making a new one lol, I included it

Staubgeborener, 2024-03-02:
btw run `git gc` in your `config_backup` folder like is said, not in the `klipper-backup` folder.

Tylerjet, 2024-03-02:
Could you try 
```cd ~
rm -rf config_backup
```
and then run the script again

Bradford1040, 2024-03-02:
cd into home? or into printer_1_data?

Tylerjet, 2024-03-02:
cd into home

Tylerjet, 2024-03-02:
what i think may be occuring is that your previous backups was using code from pre version 1.0 of klipper backup

Staubgeborener, 2024-03-02:
Depend on your setup. Normally klipper-backup is installed in home directory. But you installed it somewhere in a printer1 or printer2 folder so I'm not sure about your whole setup right now.

Bradford1040, 2024-03-02:
that did it, thank you, can't remember do I close the ticket or do you?

closed 2024-03-02 by Staubgeborener

Tylerjet, 2024-03-02:
You can close if resolved yep

Bradford1040, 2024-03-02:
might have been a KIAUH backup that messed everything up also, I'll look into that