Staubgeborener/Klipper-Backup #79: Easy way to change the service-label?

Tomblarom, 2024-03-03:
### Is your feature request related to a problem? Please describe.

I was not able to find the definition of `Klipper-backup-filewatch`. Where does this string originate from? I want to change it to `Auto-Backup` for sake of simplicity (and my ocd😅).

[image]

### Describe the solution you'd like

NOPE

### Describe alternatives you've considered

NOPE

Staubgeborener, 2024-03-03:
Do you mean how to implement the code on your system? The is a whole [section](https://staubgeborener.github.io/klipper-backup/automation/#backup-on-file-changes) in the docs for this. It's a systemd service which runs in the background. It has nothing to do with mainsail or any other UI.

Tomblarom, 2024-03-03:
No, just the naming 😁 The value of that string, beeing displayed in mainsail :)

Staubgeborener, 2024-03-03:
It's not a string which is beeing displayed in mainsail. It's just a service running in background of your system and has nothing to do with mainsail.

Or do i misunderstood something here?

Tomblarom, 2024-03-03:
Yes, you're not understanding what I mean 😁 It's just a cosmetic thing, therefore rather a feature-request. I just want to set the service-name to `Auto-Backup`, instead of `Klipper-backup-filewatch`. Like this:

[image]

Tylerjet, 2024-03-03:
The name in the UI is based on the file name you would need to rename. `Klipper-backup-filewatch.service` under `/etc/systemd/system/`

Tomblarom, 2024-03-03:
I changed it, but the result was, that the service entry disappeared from the menu. Service is running properly and doing backups. I tried this twice, with rebooting in between. Try it yourself.

```
pi@rpi3-3d-ender:~ $ sudo systemctl status auto-backup.service
[sudo] password for pi:
● auto-backup.service - Klipper Backup Filewatch Service for Auto-Backup
     Loaded: loaded (/etc/systemd/system/auto-backup.service; enabled; vendor preset: enabled)
     Active: active (running) since Sun 2024-03-03 21:11:17 CET; 2min 16s ago
   Main PID: 396 (bash)
      Tasks: 3 (limit: 719)
        CPU: 156ms
     CGroup: /system.slice/auto-backup.service
             ├─396 /bin/bash -c      watchlist="";      while IFS= read -r path; do          for file in $path; do   >
' | sort -u | tr '
' ' ');      exclude_pattern=".swp|.tmp|printer-[0-9]*_[0-9]*.cfg|.bak|.bkp";      inotifywait -mrP -e close_write -e>
             ├─431 inotifywait -mrP -e close_write -e move -e delete --exclude .swp|.tmp|printer-[0-9]*_[0-9]*.cfg|.b>
             └─432 /bin/bash -c      watchlist="";      while IFS= read -r path; do          for file in $path; do   >
' | sort -u | tr '
' ' ');      exclude_pattern=".swp|.tmp|printer-[0-9]*_[0-9]*.cfg|.bak|.bkp";      inotifywait -mrP -e close_write -e>

Mar 03 21:11:17 rpi3-3d-ender systemd[1]: Started Klipper Backup Filewatch Service for Auto-Backup.
Mar 03 21:11:18 rpi3-3d-ender bash[431]: Setting up watches.  Beware: since -r was given, this may take a while!
Mar 03 21:11:18 rpi3-3d-ender bash[431]: Watches established.
```

Tylerjet, 2024-03-03:
did you re-enable the new service? As it needs to create a new symbolic link as the original will still be linked to the old file name.
```
sudo systemctl disable klipper-backup-filewatch.service
sudo systemctl enable auto-backup.service

```

Tomblarom, 2024-03-03:
Yes... I really did that and [followed the docs](https://staubgeborener.github.io/klipper-backup/automation/#backup-on-file-changes), but used the name `Auto-Backup` instead of Klipper-backup-filewatch 😅

Tylerjet, 2024-03-03:
Hrmm not sure why it wasn't needed for the klipper-backup names but in printer_data you need to add the service to moonraker.asvc

Tomblarom, 2024-03-03:
YES! Adding `auto-backup` to `moonraker.asvc` solved it! :)

closed 2024-03-03 by Tomblarom

Tomblarom, 2024-03-03:
To summerize these are the steps I've done:
```
# Remove klipper-backup-filewatch.service
sudo systemctl stop klipper-backup-filewatch.service
sudo systemctl revert klipper-backup-filewatch.service
sudo rm /etc/systemd/system/klipper-backup-filewatch.service
sudo systemctl daemon-reload

# Recreate auto-backup.service (by following the docs)
sudo systemctl edit --force --full auto-backup.service
sudo systemctl daemon-reload
sudo systemctl enable auto-backup.service
sudo systemctl start auto-backup.service

# Add Auto-Backup to Moonraker
sed -i "$ a auto-backup" ~/printer_data/moonraker.asvc

# Reboot (important!)
sudo reboot
```