Staubgeborener/Klipper-Backup #80: Lost Network Access

stefanpieter, 2024-03-08:
### Code of Conduct

- [X] I promise that i have used markdown syntax to make the `code` more readable which helps to fix the problem

### What happened

I followed "Backup on boot" for network manager
Ran step 2 "Reload the service daemon, enable the service and start the service"
Rebooted 
No network access to the printer

### What did you expect to happen

To reboot and backup to take place.

### How to reproduce

No idea.

### Additional information

I have standard MainSailOS. All up-to-date.

Tylerjet, 2024-03-08:
What are all the files and services that you edited? The on-boot service is just set to want the network connection online and run after the network is up but it does not modify the network connection or settings.

stefanpieter, 2024-03-17:
I am not sure what happened, but all repos got corrupted. After recovering everything it is up and working again.

closed 2024-03-17 by stefanpieter