Staubgeborener/Klipper-Backup #81: Add ability to backup composed configuration that are using [included ...cfg] files.

mmemetea, 2024-03-10:
### Is your feature request related to a problem? Please describe.

My current configuration is exploded in several smaller and easier-to-manage sub-directories such as this:

#########################
### MAIN CONFIG FILES ###
#########################
[include config/machine.cfg] # Keep this included!
[include config/variables.cfg] # Keep this included!

### MCUs configuration ---------------------------------------------
[include config/mcus/SKR1.4_x2_V2.4.cfg]
[include config/mcus/rpi.cfg]
# [include config/mcus/spider_v1.cfg]
# [include config/mcus/spider_v2.cfg]
# [include config/mcus/octopus.cfg]
# [include config/mcus/canboard_SHT36-42.cfg]
# ------------------------------------------------------------------

###########################
### HARDWARE COMPONENTS ###
###########################

### XY axis configuration ------------------------------------------
[include config/hardware/XY/0.9d_xy.cfg]
# [include config/hardware/XY/1.8d_xy.cfg]
# ------------------------------------------------------------------

### Describe the solution you'd like

I want the current script to be able to fetch these sub-directories.

### Describe alternatives you've considered

None so far. I tried to modify the .env file to include those, but without success.

Tylerjet, 2024-03-10:
How are your paths setup in the .env file? For example `printer_data/config/*` would recursively grab all the files and folders that are not-symbolic links and copy them to the backup folder.

mmemetea, 2024-03-10:
I think @ManeLippert has found the reason why it doesn't work as intended. The subdirectories aren't symlinked, so it should work, but the recent change is preventing this. My installation is quite recent, so I can't know if it would have worked before.

mmemetea, 2024-03-10:
Just tried @ManeLippert suggestion. It works beautifully. Thanks.

ManeLippert, 2024-03-10:
@mmemetea No problem. Glad I could help.

Tylerjet, 2024-03-10:
ahh indeed we forgot to keep -r for recursive in thanks for the catch will place a pr here in a few moments.

as for using -a the reason for not backing up symlinks i noted here (tl:dr is sym links are read only files since they are linked to files is repositories eg.) KAMP. When KAMP is updated all those changes would be gone and reverted) https://github.com/Staubgeborener/klipper-backup/issues/69#issuecomment-1965839873

And -R is used so that the folder structure it came from is backed up, this is for future intentions of a restore function so we want to have that path in the backup. 

I also think its a bit nicer on the eyes if you choose to backup multiple folders you can see where all the files were backed up from, so a file at `printer_data/config/printer.cfg` is also at `printer_data/config/printer.cfg` in the repository.

closed 2024-03-10 by Staubgeborener

Tylerjet, 2024-03-10:
@mmemetea @ManeLippert You can now update to latest and -r is included in rsync. Thanks!

PR #82 (2024-03-10): Update script.sh
add -r (recursive) back into rsync command as it was accidentally removed when replacing old cp -r command.

Resolves #81