Staubgeborener/Klipper-Backup #83: [Bug]:

3DeezNutz, 2024-03-12:
### Code of Conduct

- [X] I promise that i have used markdown syntax to make the `code` more readable which helps to fix the problem

### What happened

I tried updating and im getting this error. 
[image]

### What did you expect to happen

just to update my back up files.

### How to reproduce

just hit the gitupdate macro.

### Additional information

[image]

Tylerjet, 2024-03-12:
I had this happen but not really sure why it occurs what you can try is going into the config_backup folder via ssh and running `git pull` and i believe that should be able to un corrupt the repo

3DeezNutz, 2024-03-12:
What do you mean by git pull? I’m not very good at this.

Tylerjet, 2024-03-12:
git pull is a command, once logged into the PC running klipper you will want to do:
```
cd config_backup
```
and then ```
git pull
```

and see what errors occur if any

3DeezNutz, 2024-03-12:
[image]

like this?

Tylerjet, 2024-03-12:
yep ok so try this instead
```
git init
```
and then 
```
git pull
```

3DeezNutz, 2024-03-12:
[image]

3DeezNutz, 2024-03-12:
[image]

Tylerjet, 2024-03-12:
could you try a ```git pull``` again what shows up

3DeezNutz, 2024-03-12:
ok thank you that fixed it

3DeezNutz, 2024-03-12:
how would i go about backing up my history on mainsail?

Tylerjet, 2024-03-12:
Not sure what you mean are you referring to the the mainsail.cfg file? that is a read only file and can't be modified in the editor so it is skipped from backups.

3DeezNutz, 2024-03-12:
like all the data from your total print hours. 
[image]

Tylerjet, 2024-03-12:
That data is saved in a database specific to moonraker and has a bit more involved process to backup the data base to be able to restore it if needed later.

3DeezNutz, 2024-03-12:
Could this be a future request feature?

Staubgeborener, 2024-03-13:
Maybe, it's possible to [backup and restore](https://moonraker.readthedocs.io/en/latest/installation/#lmdb-database-backup-and-restore) the moonraker database. We could add a simple `~/moonraker/scripts/backup-database.sh -o ~/moonraker-database.backup` command line in future ([code](https://github.com/Arksine/moonraker/blob/master/scripts/backup-database.sh)) which can be toggled with a `backup_database` parameter (or something like that) in future.
Right now you can backup this database by yourself and add the path to the `data.mdb` file in your `.env`.

3DeezNutz, 2024-03-13:
Do I use single file?

Staubgeborener, 2024-03-13:
What.

3DeezNutz, 2024-03-13:
do i use this 
[image]

Staubgeborener, 2024-03-13:
Yes. If you run the backup like this
`~/moonraker/scripts/backup-database.sh -o ~/moonraker-database.backup`
You create the backup file of your database in your home directory. 

So basically adding something like this in your `.env` is enough:
`path_moonrakerdatabase=moonraker-database.backup`

Staubgeborener, 2024-03-13:
Yes. If you run the backup like this
`~/moonraker/scripts/backup-database.sh -o ~/moonraker-database.backup`
You create the backup file of your database in your home directory. 

So basically adding something like this in your `.env` is enough:
`path_moonrakerdatabase=moonraker-database.backup`

I would suggest to create the database backup in a extra directory and not in your home directory. After pushing all files to GitHub so you avoid disorganization in your repository.

closed 2024-04-11 by github-actions[bot]