Staubgeborener/Klipper-Backup #88: Security issue - telegram bot conf

LegionPleingaz, 2024-03-22:
### Is your feature request related to a problem? Please describe.

With the global push to a git, the bot creds are exposed.
is it possible to exclude one to x files?

### Describe the solution you'd like

Modify in the .env file

### Describe alternatives you've considered

no idea

Tylerjet, 2024-03-22:
You don't have to have your backup repository public but you can exclude files by adding them to the exclude list in the .env

[image]

Staubgeborener, 2024-03-22:
Also please read [this](https://staubgeborener.github.io/klipper-backup/configuration/#gitignore-do-not-upload-certain-files).

closed 2024-04-20 by github-actions[bot]