Staubgeborener/Klipper-Backup #89: Add Restore Script

Salty-Doug, 2024-03-25:
### Is your feature request related to a problem? Please describe.

No.

### Describe the solution you'd like

It would be nice to have a restore script that would restore all files.

### Describe alternatives you've considered

I am not a Github developer.

Staubgeborener, 2024-03-25:
Is being planned and will be released shortly in some weeks/months. The current focus is on the installation script and KIAUH support.

closed 2024-03-25 by Staubgeborener

Salty-Doug, 2024-03-26:
Thanks for the reply.I found that the Installation went Pretty good between the video and your documentation. Fair Winds and Calm SeasSalty Doug SimmsSent from my iPhone