Staubgeborener/Klipper-Backup #90: Does not back up symbolic links like those used by ERCF

jwhite, 2024-03-28:
### Code of Conduct

- [X] I promise that i have used markdown syntax to make the `code` more readable which helps to fix the problem

### What happened

By default, this will skip symbolic links, which is probably not the desired behavior.

Adding the -l flag to rsync will copy those.

### What did you expect to happen

My backups would be created.

### How to reproduce

Install any klipper extension that puts symbolic links into printer configuration.

Tylerjet, 2024-03-28:
Symbolic Links are not backed up as they are read only files https://github.com/Staubgeborener/klipper-backup/issues/69#issuecomment-1965839873

Tylerjet, 2024-03-28:
You can see it in Fluidd and mainsail denoted by the lock icon and in some of the files denoting they are read only.

[image]
[image]

closed 2024-04-14 by Tylerjet