Staubgeborener/Klipper-Backup #95: add bigger http.postBuffer to script.sh

fabiandj64, 2024-05-02:
### Is your feature request related to a problem? Please describe.

i had the problem that i could not push big gcode files to git because it filled the postBuffer. i had to manually add "git config http.postBuffer 524288000" right before the git push command in the script.sh file. maybe it could be implemented that u can upload larger files

### Describe the solution you'd like

add "git config http.postBuffer 524288000" right before the git push command in the script.sh file.

### Describe alternatives you've considered

use git large file system as a workaround for big files [git lfs](https://git-lfs.com/)

Tylerjet, 2024-05-02:
That is likely a bit out of scope for this, klipper backup is meant to assist in backing up your config and not gcode files.

fabiandj64, 2024-05-02:
Alright I thought it could help others with uploading larger files. But I guess if u really need it u could edit the file yourself and add the buffer command

closed 2024-05-02 by fabiandj64

celulari, 2024-05-19:
> ### Is your feature request related to a problem? Please describe.
> i had the problem that i could not push big gcode files to git because it filled the postBuffer. i had to manually add "git config [...]

Thanks for the info
I was getting erros on updating because i'm uploading my gcodes