Staubgeborener/Klipper-Backup #96: [Feature request]: Check that the printer is not running before starting the backup

Pelado-Mat, 2024-05-02:
### Is your feature request related to a problem? Please describe.

I got failed prints on my setup, all with a "MCU overloaded" error exactly at the time of the scheduled backup

### Describe the solution you'd like

add to the backup script an, perhaps optional, check to skip the backup if the printer is running

```
# abort if its printing

STATE=$(curl -X GET http://127.0.0.1:7125/printer/objects/query?print_stats 2>/dev/null | grep -oP '(?<="state": ")[^"]*')
if [ "$STATE" = "printing" ]; then
  exit 0
fi
```

### Describe alternatives you've considered

Perhaps moving to a newer rPI, but is working fine besides the backup.

Tylerjet, 2024-05-02:
Huh that is interesting what device are you currently running Klipper on as even on a neptune 4 pro which doesn't have the best clone of a mkspi doesn't seem to have issues. 

The cron job is also an optional feature and can be modified without affecting the klipper backup repository and causing it to report as dirty. so you could just add that code into a script file that is then called from the cron or directly to the cron command.

Pelado-Mat, 2024-05-02:
> Huh that is interesting what device are you currently running Klipper on as even on a neptune 4 pro which doesn't have the best clone of a mkspi doesn't seem to have issues.
>
> The cron job is also an [...]

It's an old rPI 3+ with a dubious SD Card. I have also the rPI driving a Oled screen and keypad using a the rpi as a klipper mcu, perhaps that why is falling.

I'm running that code on my backup script, it working fine.

Tylerjet, 2024-05-02:
Gotcha, What I meant was instead of it being placed in the backup script itself where it runs fine already for others without it, this makes more sense to modify the optional code of the cron than making it a permanent change to the main.

closed 2024-05-31 by github-actions[bot]