Staubgeborener/Klipper-Backup #98: Pushing updates doesn't work

Settordici, 2024-05-31:
### Code of Conduct

- [X] I promise that i have used markdown syntax to make the `code` more readable which helps to fix the problem

### What happened

I ran `~/klipper-backup/script.sh`, and it didn't push to the repository. I attached a screenshot of the terminal output

### What did you expect to happen

I expected it to push the updates

### How to reproduce

Just run `~/klipper-backup/script.sh`, but it's probably something that happened on my end

### Additional information

[image]

Tylerjet, 2024-05-31:
can you try removing the config backup folder and then try running script.sh again:
```
cd
rm -rf config_backup
```

also is this an existing repo or a brand new one created for use with the backup script?

Settordici, 2024-05-31:
It worked, thank you for you quick answer.

I created the repo first, and then ran the script

Tylerjet, 2024-05-31:
No Problem!

closed 2024-05-31 by Tylerjet