Staubgeborener/Klipper-Backup #99: backup not working with BTT Pad 7

bulcke, 2024-06-02:
### Code of Conduct

- [X] I promise that i have used markdown syntax to make the `code` more readable which helps to fix the problem

### What happened

I'm using a BTT Pad 7 with CB1. It's fully updated.
Using script.sh, the backup fails. Full output:

`biqu@BTT-Pad7:~/klipper-backup$ ./script.sh
● Checking for installed dependencies Done!

Klipper-backup is up to date

hint: Pulling without specifying how to reconcile divergent branches is
hint: discouraged. You can squelch this message by running one of the following
hint: commands sometime before your next pull:
hint:
hint:   git config pull.rebase false  # merge (the default strategy)
hint:   git config pull.rebase true   # rebase
hint:   git config pull.ff only       # fast-forward only
hint:
hint: You can replace "git config" with "git config --global" to set a default
hint: preference for all repositories. You can also pass --rebase, --no-rebase,
hint: or --ff-only on the command line to override the configured default per
hint: invocation.
error: index uses �9n� extension, which we do not understand
fatal: index file corrupt
Skipping symbolic link: printer_data/config/mainsail.cfg
Skipping symbolic link: printer_data/config/print_area_bed_mesh.cfg
Skipping symbolic link: printer_data/config/timelapse.cfg
error: index uses �9n� extension, which we do not understand
fatal: index file corrupt
error: index uses �9n� extension, which we do not understand
fatal: index file corrupt
error: index uses �9n� extension, which we do not understand
fatal: index file corrupt
Branch 'main' set up to track remote branch 'main' from 'origin'.
Everything up-to-date`

I don't know what that means.
My configs are not updated to the github repo.

### What did you expect to happen

Configs to be uploaded to my github repository.

### How to reproduce

Use fully up to date BTT Pad 7 with CB1 and new installation of klipper-backup.
run ./script.sh

Tylerjet, 2024-06-02:
Is the repo you are pushing to brand new or does it have previous files from a different backup service/manual backups? 

If you delete the config backup folder:
```
cd
rm -rf config_backup
```
And then ensure the remote repo is completely empty and run script.sh does it run into errors?

bulcke, 2024-06-02:
> Is the repo you are pushing to brand new or does it have previous files from a different backup service/manual backups?
> 
> If you delete the config backup folder:
> 
> ```
> cd
> rm -rf config_backup
> ```
> 
> And then ensure the remote repo is completely empty and run script.sh does it run into errors?

Since I was having issues, I retried with a new repo and I reinstalled/reconfigured klipper-backup. But I failed to remove the config_backup directory. Working now. Thank you

For a simple user, having basic knowledge of what the script does in what folders, may be of use. I had no idea this folder is part of klipper-backup. The config_backup folder is only mentioned once in the docs (https://klipperbackup.xyz/configuration/#gitignore-do-not-upload-certain-files).

Thank's again for the swift reply and keep up the good work!

closed 2024-06-02 by bulcke

Tylerjet, 2024-06-03:
Yeah we could probably add a troubleshooting page at some point, though some of the items it would be nice to have the script catch and either self resolve or just notify the steps needed to resolve.

This one could maybe be a self resolve of if the backup fails from git errors we just delete the folder and try again without needing user intervention.

Staubgeborener, 2024-06-03:
> Yeah we could probably add a troubleshooting page

Definitive. I also wanted to embed more gifs in the future like [this one](https://imgur.com/kOct70v) anyway to show animated tutorials (now i just need time for something like this).