Staubgeborener/Klipper-Backup #100: Expand Backed Up Files

mlee12382, 2024-06-04:
### Is your feature request related to a problem? Please describe.

Currently only the config folder is backed up.

### Describe the solution you'd like

Change the backup to include all of the printer_data folder and not just printer_data/config so that printer history and log files are also included in the backup at a minimum. Add the timelapse folder, if there is one, to the ignore list to reduce the space needed on github.

### Describe alternatives you've considered

NA

Tylerjet, 2024-06-04:
This can already be done.

The .env.example file explains it but will also paste here as well.
```
# All information regarding .env can be found here:
# https://klipperbackup.xyz/configuration/

# Backup paths
# Note: script.sh starts its search in $HOME which is /home/{username}/
# The array accepts folders or files like the following example
# 
#  backupPaths=( \
#  "printer_data/config/*" \
#  "printer_data/config/printer.cfg" \
#  )
#
# Using the above example the script will search for `/home/{username}/printer_data/config/*` and `/home/{username}/printer_data/config/printer.cfg`
# When backing up a folder you should always have `/*` at the end of the path so that files insde the folder are properly searched 

backupPaths=( \
"printer_data/config/*" \
)
```

You would just need to to make the backup path `printer_data/*` instead of `printer_data/config/*`

mlee12382, 2024-06-04:
> This can already be done.
>
> The .env.example file explains it but will also paste here as well.
>
> ```
> # All information regarding .env can be found here:
> # https://klipperbackup.xyz/configuration/
>
> # [...]

Correct, this should be the default though for the project and not have to be modified by the end user. Remember not everyone has the knowledge or skills to do more than just copy and paste based on instructions. There's no downside to backing up more than is currently being done and there's plenty of upside. It should at least be an option that is easily changed when running the install script if nothing else.

Staubgeborener, 2024-06-04:
All you have to do is enter the paths and files you want to have in your backup. No more or less, we can hardly make it simpler. 

Otherwise, simply consider the content of the .env as an example of the syntax if it does not suit your needs. For example, not everyone wants to have logfiles in their backup, but wants to keep the backup minimalistic with the basic configuration.

Small example: if we change `printer_data/config/*` to `printer_data/*` by default, your "problem" would be solved, but a lot of issues would be opened here that would then say "why is so much nonsense included in the backup, I only need my config". Everything has two sides. If you want to include more -> simply adapt .env. We can't reproduce every case here, because the next user will post an issue with "why don't you include this random plugin which is used by 5 people in the backup by default".

mlee12382, 2024-06-04:
> All you have to do is enter the paths and files you want to have in your backup. No more or less, we can hardly make it simpler.
>
> Otherwise, simply consider the content of the .env as an example of [...]

I understand that, I still think it would be a good idea to at least add a prompt in the install script for minimalistic which would be the current config file only, full which would be the whole printer_data folder or custom which would be user specified. All of that could be easily handled in the script which would make it a lot easier for the increasing majority of klipper users that don't know how to do anything beyond copying and pasting based on written instructions or YouTube tutorials.

Staubgeborener, 2024-06-04:
If we would include a "full which would be the whole printer_data folder" backup feature in the installation script, we would get questions and issues why the plugin xyz (which is not listed under `printer_data`, but in the home directory for example) is not included when it is a “full backup”. Trust me, after this project i can say, that it is very obvious where some users will get more problems than solutions.

closed 2024-06-04 by Staubgeborener