Staubgeborener/Klipper-Backup #101: Allow deploy tokens in addition to access tokens

PhilippMolitor, 2024-06-07:
### Is your feature request related to a problem? Please describe.

Currently, only access tokens are supported. The new tokens have a one year lifetime, so they must be replaced.

### Describe the solution you'd like

With deploy tokens (SSH keys), one could limit the access of the token to only the backup repo, while still having indefinite lifetime. Bonus: they can be generated by the installer on the target system with `ssh-keygen`!

### Describe alternatives you've considered

-

Tylerjet, 2024-06-07:
Definitely something i want to do, I use ssh-keys way more myself as well. Likely wouldn't be in the installer script though more likely would just be a config flag in the .env just like empty_commits that if set to true it ignores the token field and connects via ssh.

PhilippMolitor, 2024-06-20:
So this is working via #103 .

I still think it would be in the interest of the users to add an optional ssh-keygen step to the installer. i imagine something like:
```
would you generate and set up an SSH deploy token? (recommended) [Y/n]: Y
creating ssh key...
$ ssh-keygen -q -t rsa -b 4096 -N '' -f ~/.ssh/id_rsa <<<y >/dev/null 2>&1
ssh key created! your deploy token is:

ssh-rsa AAAAB3NzaC1yc2EAAAA ...

Please enter this token at https://github.com/$github_user/$github_repo/settings/keys
```

It should ofc check if the id_rsa file is already present, or maybe even use a custom name by default.
I think this is the most convenient way to set up klipper-backup by default, as this does not require any token creation at GitHub, the installer can do that by itself and then outputs the pubkey to copy/paste to the form at the given URL.

Tylerjet, 2024-07-03:
Will close for now as the original request was complete but will still think about if we may want to add this into the isntaller or keep as a more advanced setup with some notes adding into the wiki.

closed 2024-07-03 by Tylerjet