Staubgeborener/Klipper-Backup #105: Moonraker backups

PhilippMolitor, 2024-06-20:
### Is your feature request related to a problem? Please describe.

Moonraker has a built-in tool for creating database backups:
https://moonraker.readthedocs.io/en/latest/installation/?h=backup#database-backup-and-restore

It seems to create base64 encoded backups of the database.

### Describe the solution you'd like

klipper-backup should be able to leverage the moonraker backup tool to create db backups. this would also be beneficial for the planned restore function.

### Describe alternatives you've considered

I am currently using this config:

```env
allow_empty_commits="false"

backupPaths=( \
"printer_data/config/*" \
"printer_data/database/*" \
)
```

Problems:
- it creates a lot of empty commits because the DB file is written to very often, this can be circumvented with `allow_empty_commits=false`
- the db files are binary, which clutters the git repo with non-diffable data.
- if the database file is currently written to, it could result in a corrupt backup

Tylerjet, 2024-06-29:
This is something i looked at a while back but it can somtimes be a pain if i recall correctly, so will possibly look back into it in the future.

PhilippMolitor, 2024-07-14:
bump @ stale

Tylerjet, 2024-07-16:
Added commit to restore-dev that will allow for a variable ```moonraker_db_backups``` to be set true/false that will run moonrakers backup script and will restore the database file if found on restore actions.

PhilippMolitor, 2024-08-14:
please open again :)

Tylerjet, 2024-08-29:
.

Tylerjet, 2025-03-21:
@PhilippMolitor if you have a testing environment #143

Tylerjet, 2025-04-01:
Will be resolved by #143 once merged after beta period.

closed 2025-04-01 by Tylerjet