Staubgeborener/Klipper-Backup #106: mmu_vars.cfg exclude from filewatch

mrmultifunk, 2024-06-27:
### Is your feature request related to a problem? Please describe.

I have a ERCF on my voron, running the Happy Hare software. The software uses a var file, mmu_vars.cfg that stores the variables for the Happy Hare software - this files changes a lot during printing, but is useful to backup manually.

### Describe the solution you'd like

Can that file be excluded from the filewatch script? Eg.

exclude_pattern=".swp|.tmp|printer-[0-9]*_[0-9]*.cfg|.bak|.bkp|mmu_vars.cfg"

### Describe alternatives you've considered

The file could be added to the exclude list in the .env file, but a like to have it backed up manually.

Tylerjet, 2024-06-29:
Hrmm I will have to look into allowing modifications of the filewatch service excludes variable maybe having its own entry in the .env.

With the current implementation since the service is manually created, you can edit the exclude line of the service to have the mmu_vars.cfg while leaving it out of the main .env so that the watcher will not attach to the mmu_vars.cfg file.

mrmultifunk, 2024-06-29:
Thanks for looking into it. I did add the mmu_vars.cfg to the script, and for now it seems to work.

Tylerjet, 2024-07-08:
Made the changes on the restore-dev branch (still unstable-ish) but will make that functionality i mentioned available in a future release.

closed 2024-07-08 by Tylerjet