Staubgeborener/Klipper-Backup #107: Use systemd timers instead of cron

whezzel, 2024-07-02:
### Is your feature request related to a problem? Please describe.

systemd already has support for timers removing the requirement to install cron.

### Describe the solution you'd like

Create a timer unit file at `/etc/systemd/system/klipper-backup.time`
```bash
# /etc/systemd/system/klipper-backup.timer
[Unit]
Description=Run klipper backup script every 4 hours

[Timer]
OnBootSec=300s
OnUnitActiveSec=4hr

[Install]
WantedBy=timers.target
```

Create a service file at `/etc/systemd/system/klipper-backup.service`
```bash
# /etc/systemd/system/klipper-backup.service
[Unit]
Description=Klipper Backup On-boot Service
#Uncomment below lines if using network manager
After=NetworkManager-wait-online.service
Wants=NetworkManager-wait-online.service
#Uncomment below lines if not using network manager
#After=network-online.target
#Wants=network-online.target

[Service]
User=<REPLACE_WITH_YOUR_USERNAME>
Type=oneshot
ExecStart=/bin/bash -c 'bash $HOME/klipper-backup/script.sh "New Backup on timer - $(date +"%%x - %%X")"'

[Install]
WantedBy=default.target
```

Once both files are created, run `systemctl daemon-reload` then enable the systemd timer `systemctl enable --now klipper-backup.timer`. The timer will run the service file every 4 hours.

### Describe alternatives you've considered

I tried the cron option, but I don't see the need to install cron when systemd already supports timers.

Tylerjet, 2024-07-02:
I'm pretty sure Cron doesn't need to be installed as it should be standard in linux installs? however the services and cron are optional so you are able to implement your own setup if you so choose.

whezzel, 2024-07-02:
It depends on the distro. The distro I primarily use doesn't have cron installed by default. I did implement this manually, though I thought I would mention it here as another option, especially since this already uses systemd service units for running the script.

I'll close this if there is no interest in officially implementing this.

Tylerjet, 2024-07-02:
Yeah I can make a note to add it in the wiki docs as an alternative method, out of curiosity what distro are you running that cron is not included.

whezzel, 2024-07-02:
I use arch. cron is in the repos, just not included in the base packages.

Tylerjet, 2024-07-03:
Added new section to docs referencing the above setting with a note about it in automation under the cron section.

closed 2024-07-03 by Tylerjet

whezzel, 2024-07-12:
Sorry to reopen this. I just noticed there is a typo. In step 1, it says this `Create a timer unit file at /etc/systemd/system/klipper-backup.time` but it should be `Create a timer unit file at /etc/systemd/system/klipper-backup.timer`

Tylerjet, 2024-07-12:
Fixed

Staubgeborener, 2024-07-13:
@whezzel thanks, all in all we are very open for pull request. So if you see anything which can be improved in the docs, just switch to this branch under open a PR. Again: thanks for the hint!