Staubgeborener/Klipper-Backup #110: submodule error

therick0996, 2024-07-15:
when running the script, i get this:

```
warning: adding embedded git repository: printer_data/config
hint: You've added another git repository inside your current repository.
hint: Clones of the outer repository will not contain the contents of
hint: the embedded repository and will not know how to obtain it.
hint: If you meant to add a submodule, use:
hint: 
hint: 	git submodule add <url> printer_data/config
hint: 
hint: If you added this path by mistake, you can remove it from the
hint: index with:
hint: 
hint: 	git rm --cached printer_data/config
hint: 
hint: See "git help submodule" for more information.
```

i've tried running `git rm --cached printer_data/config` and get 
```
fatal: not a git repository (or any of the parent directories): .git
```
i've also tried deleting all klipper-backup files from the pi and reinstalling

Grantsauto, 2024-07-15:
I have also run into this recently, all my git files are locked and unavailable. I have removed and reinstalled multiple times with the same error and in a new repository. Skips symlink for all files.

Tylerjet, 2024-07-16:
Are you using a klipper config from a git repo, Where you have cloned that repos contents into printer_data/config?

If so you navigate to the config folder and run 'git remote get-url original' what is returned.

therick0996, 2024-07-16:
@Tylerjet aha! i had previously installed a different git backup thingy

`https://<git_token>@github.com/therick0996/v2_config.git`

i tried deleting `v2_config` repo, and still get the error and result from `git remote get-url origin`. can you help an idiot like me and remove this?

therick0996, 2024-07-16:
thank you so much!

closed 2024-07-16 by therick0996

Staubgeborener, 2024-07-16:
> you will need to delete the .git folder inside of printer_data/config and that should fix it.

We should add this in the docs or maybe implement this workaround in the script itself: Run script -> .git folder error -> delete folder -> run script