Staubgeborener/Klipper-Backup #112: backup error

qp1306, 2024-07-31:
I'm getting this error when trying to do a back up and I don't kn ow what it means

11:34 AM
Command {update_git_script} finished
11:34 AM
fatal: the remote end hung up unexpectedly
error: failed to push some refs to 'https://github.com/qp1306/VzBot235_Goliath.git'
11:34 AM
send-pack: unexpected disconnect while reading sideband packet
11:34 AM
fatal: the remote end hung up unexpectedly
11:34 AM
fatal: loose object 73dd169d185e33b3e7e4c5de5effe4a77d3965ce (stored in .git/objects/73/dd169d185e33b3e7e4c5de5effe4a77d3965ce) is corrupt
11:34 AM
error: object file .git/objects/73/dd169d185e33b3e7e4c5de5effe4a77d3965ce is empty
error: object file .git/objects/73/dd169d185e33b3e7e4c5de5effe4a77d3965ce is empty
11:34 AM
error: object file .git/objects/73/dd169d185e33b3e7e4c5de5effe4a77d3965ce is empty
error: object file .git/objects/73/dd169d185e33b3e7e4c5de5effe4a77d3965ce is empty
fatal: loose object 73dd169d185e33b3e7e4c5de5effe4a77d3965ce (stored in .git/objects/73/dd169d185e33b3e7e4c5de5effe4a77d3965ce) is corrupt
11:34 AM
Skipping symbolic link: printer_data/config/mainsail.cfg
11:34 AM
Klipper-backup is up to date
11:34 AM
[1;93m●[0m Checking for installed dependencies /
[K[1;92m●[0m Checking for installed dependencies [1;92mDone![0m
11:34 AM
Running Command {update_git_script}...:

Tylerjet, 2024-07-31:
can you try the following:
https://github.com/Staubgeborener/klipper-backup/issues/77#issuecomment-1974929037

qp1306, 2024-07-31:
Thanks that has fixed my issue.

Staubgeborener, 2024-07-31:
This is now a [parameter](https://klipperbackup.xyz/#troubleshooting).

closed 2024-07-31 by Staubgeborener