Staubgeborener/Klipper-Backup #114: Klipper-back up stooped working

Jaizu, 2024-08-05:
### Code of Conduct

- [X] I promise that i have used markdown syntax to make the `code` more readable which helps to fix the problem and that I have tested the `--fix` command mentioned above without success.

### What happened

Klipper-backup stopped working a few days ago.

### What did you expect to happen

It workings

### How to reproduce

I tried restarting my btt pi to see if it makes a backup, it doesnt. It also doesnt backup every X time.

### Additional information

[image]

Staubgeborener, 2024-08-05:
Did you update to the current version a few days ago? I only added the possibility for parameters there; actually there should only be problems if you wanted to use individual commit messages (old: `script.sh "my commit"` / new: `script.sh -commit_message "my commit"`). Did you use this before? 

Otherwise I would need to see your `.env` file (hide the token) and an exact error message what happens when you run `script.sh` (like screenshots).

Staubgeborener, 2024-08-09:
Thanks for pointing that out, I actually forgot something. Could you please update to the latest version and confirm that everything is working? It should have been affected: filewatch service, backup on boot cron, "every 4 hours" cron

crkochan, 2024-08-09:
Filewatch service is confimed as fixed.

Jaizu, 2024-08-09:
I updated klipper-backup, restarted my raspberry pi, restarted klipper, restarted Klipper-backup-filewatch and Klipper-backup-on-boot and I still see no changes updated to my repo.
[image]
I will check later today again to see if something changed.

Staubgeborener, 2024-08-09:
@Jaizu Please check in your local files if `-c` is present in your [filewatch.sh](https://github.com/Staubgeborener/klipper-backup/blob/46321046222c1a707873a24eb18b9ea5377cf815/utils/filewatch.sh#L28) file.

I think you have to re-install the script. Create a backup of your `.env` file, and run the `install.sh` file. Input the relevant lines from your `.env` backup while running the installation and in this setup reinstall the filewatch and cronjob automatically. It will then automatically pull the corrected code lines.

Staubgeborener, 2024-08-09:
About step 1:
Screenshot looks good, filewatch service should work. Did you create a test file `touch ~/printer_data/config/testfile` to check if the service is truly not working? And is the service itself truly running `sudo systemctl status klipper-backup-filewatch.service`?

About step 2:
> I did step 2 and nano opened a blank file

Which means you never used this "backup every 4 hours" cron job. Otherwise you would have an entry, so i guess you are only using filewatch and backup-on-boot.

About step 3:
Did you received a error message running `sudo cp ~/klipper-backup/install-files/klipper-backup-on-boot.service /etc/systemd/system/klipper-backup-on-boot.service`? Please show the output of `sudo cat /etc/systemd/system/klipper-backup-on-boot.service` and `sudo systemctl status klipper-backup-on-boot.service`.

Jaizu, 2024-08-09:
About step 1:
The service is running.
[image]
Do you want me to run `touch ~/printer_data/config/testfile` anyways?
I did enabled the backup every 4 hours when I installed it tho.

About step 2:
```
biqu@porygon:~$ crontab -l
biqu@porygon:~$ sudo crontab -l
no crontab for root
biqu@porygon:~$
```

About step 3:
[image]
And
[image]

Staubgeborener, 2024-08-09:
About step 1 and 2:
Please try the `touch` command, otherwise rerun the `install.sh` script to reinstall the backup scripts

About step2:
Please run `sudo systemctl daemon-reload` like it's explained in the error message. Otherwise run `sudo systemctl restart klipper-backup-on-boot.service`.

Staubgeborener, 2024-08-12:
I have worked a bit on the `--fix` parameter to cover cases like this. 

Can you please test the following: Save the following code in a file `fix.sh`. Make it executable with `chmod +x fix.sh` and execute it with `./fix.sh`. It is currently still somewhat experimental, but the following is checked:

- Is the config_backup folder there? If yes, delete it
- Is the cronjob (backup every 4 hours) correct? If no, adjust
- Is the clipper backup on boot service correct? If no, adjust
- Check if there is a clipper macro somewhere in `~/printer_data/config/`. If yes, point out to the user that he may have to change it himself

```shell
loading_wheel() {
    local frames="/ - \\ |"
    local delay=0.1

    while :; do
        for frame in $frames; do
            echo -n -e "\r$1 $frame"
            sleep $delay
        done
    done
}

getcursor() {
    local pos
    IFS='[;' read -p $'\e[6n' -d R -a pos -rs || echo "failed with error: $? ; ${pos[*]}"
    echo "${pos[1]}"
}
# Move cursor up one line and clear the line
clearUp() {
    echo -e "\r\033[K\033[1A"
}

R=$'\e[1;91m' # Red ${R}
G=$'\e[1;92m' # Green ${G}
Y=$'\e[1;93m' # Yellow ${Y}
M=$'\e[1;95m' # Magenta ${M}
C=$'\e[96m'   # Cyan ${C}
NC=$'\e[0m'   # No Color ${NC}

ask_yn() {
    while true; do
        read -rp "$1 (yes/no, default is yes): " answer
        case $answer in
        [Yy]* | "") return 0 ;;
        [Nn]*) return 1 ;;
        *) ;;
        esac
    done
}

fix(){
    sudo -v
    # Delete config_backup directory
    pos1=$(getcursor)
    loading_wheel "${Y}●${NC} Delete config_backup folder" &
    loading_pid=$!
    if [ ! -d "$HOME/$backup_folder" ]; then
        echo -e "${R}Nothing to delete: config_backup folder does exist${NC}\n"
    else
        rm -rf ~/config_backup
        tput cup $((pos1 - 1)) 0
        tput ed
        echo -e "${G}●${NC} Delete config_backup folder ${G}Done!${NC}\n"
    fi
    kill $loading_pid

    # Check and fix klipper-backup-on-boot.service
    if [[ "$(md5sum $HOME/klipper-backup/install-files/klipper-backup-on-boot.service | awk '{print $1}')" != "$(md5sum /etc/systemd/system/klipper-backup-on-boot.service | awk '{print $1}')" ]]; then
        pos1=$(getcursor)
        if ask_yn "klipper-backup-on-boot.service is corrupted, should this be fixed? ${R}Attention: the klipper-backup-on-boot.service will be overwritten with the klipper-backup default settings, manual changes to this file will be overwritten!${NC}"; then
            loading_wheel "${Y}●${NC} Fix \"backup on boot\" service" &
            loading_pid=$!
            sudo cp ~/klipper-backup/install-files/klipper-backup-on-boot.service /etc/systemd/system/klipper-backup-on-boot.service
            sleep .5
            kill $loading_pid
            tput cup $((pos1 - 1)) 0
            tput ed
            echo -e "${G}●${NC} Fix \"backup on boot\" service ${G}Done!${NC}\n"
        fi
    else
        echo -e "${G}●${NC} Check \"backup on boot\" service ${G}Done!${NC}\n"
    fi

    # Check and fix "automatic backup every 4 hours" cronjob
    # Check if Klipper-Backup is present in cron list
    if crontab -l | grep -q '/klipper-backup/script.sh'; then
        # If yes, check if the command is correct
        if ! crontab -l | grep -qF '0 */4 * * * /home/kubuntu/klipper-backup/script.sh -c "Cron backup - $(date +"%x - %X")"'; then
            if ask_yn "The \"automatic backup every 4 hours\" cronjob is faulty. It is: $(crontab -l | grep '/klipper-backup/script.sh'). Klipper-Backup uses the following variant: 0 */4 * * * \$HOME/klipper-backup/script.sh -c \"Cron backup - \$(date +\"%x - %X\")\". Should it be corrected? ${R}Attention: This function is experimental depending on the cron list. It is always recommended to edit the list manually${NC}"; then
                loading_wheel "${Y}●${NC} Fix \"automatic backup every 4 hours\" cronjob" &
                loading_pid=$!
                (crontab -l | grep -v '/klipper-backup/script.sh'; echo '0 */4 * * * /home/kubuntu/klipper-backup/script.sh -c "Cron backup - $(date +"%x - %X")"') | crontab -
                sleep .5
                kill $loading_pid
                tput cup $((pos1 - 1)) 0
                tput ed
                echo -e "${G}●${NC} Fix \"automatic backup every 4 hours\" cronjob ${G}Done!${NC}\n"
            fi
        else
            echo -e "${G}●${NC} Check \"automatic backup every 4 hours\" cronjob ${G}Done!${NC}\n"
        fi
    else
        echo -e "${G}●${NC} Check \"automatic backup every 4 hours\" cronjob ${G}Done!${NC}\n"
    fi

    # Check Klipper macro
    simple_klipper_macro_pattern='/klipper-backup/script.sh'
    klipper_macro_pattern='[gcode_macro update_git]
    gcode:
        {% set message = params.MESSAGE|default() %}
        {% if message %}
            RUN_SHELL_COMMAND CMD=update_git_script_message PARAMS="'{params.MESSAGE}'"
        {% else %}
            RUN_SHELL_COMMAND CMD=update_git_script
        {% endif %}

    [gcode_shell_command update_git_script]
    command: bash -c "bash $HOME/klipper-backup/script.sh"
    timeout: 90.0
    verbose: True

    [gcode_shell_command update_git_script_message]
    command: bash -c "bash $HOME/klipper-backup/script.sh -c $0"
    timeout: 90.0
    verbose: True'

    if grep -rqF "$simple_klipper_macro_pattern" "$HOME/printer_data/config/"; then
        echo "Klipper macro pattern found in the following files:"
        grep -rlF "$simple_klipper_macro_pattern" "$HOME/printer_data/config/"
        echo -e "\nPlease check yourself whether the structure is correct. The structure is expected as follows:\n"
        echo -e "$klipper_macro_pattern"
    fi

    exit
}

fix
```

Staubgeborener, 2024-08-13:
@Jaizu 

I have worked a bit on the `--fix` parameter to cover cases like this. 

Can you please test the following: Save the following code in a file `fix.sh`. Make it executable with `chmod +x fix.sh` and execute it with `./fix.sh`. It is currently still somewhat experimental, but the following is checked:

- Is the config_backup folder there? If yes, delete it
- Is the cronjob (backup every 4 hours) correct? If no, adjust
- Is the clipper backup on boot service correct? If no, adjust
- Check if there is a clipper macro somewhere in `~/printer_data/config/`. If yes, point out to the user that he may have to change it himself
- Reload systemd 

```shell
loading_wheel() {
    local frames="/ - \\ |"
    local delay=0.1

    while :; do
        for frame in $frames; do
            echo -n -e "\r$1 $frame"
            sleep $delay
        done
    done
}

getcursor() {
    local pos
    IFS='[;' read -p $'\e[6n' -d R -a pos -rs || echo "failed with error: $? ; ${pos[*]}"
    echo "${pos[1]}"
}
# Move cursor up one line and clear the line
clearUp() {
    echo -e "\r\033[K\033[1A"
}

R=$'\e[1;91m' # Red ${R}
G=$'\e[1;92m' # Green ${G}
Y=$'\e[1;93m' # Yellow ${Y}
M=$'\e[1;95m' # Magenta ${M}
C=$'\e[96m'   # Cyan ${C}
NC=$'\e[0m'   # No Color ${NC}

ask_yn() {
    while true; do
        read -rp "$1 (yes/no, default is yes): " answer
        case $answer in
        [Yy]* | "") return 0 ;;
        [Nn]*) return 1 ;;
        *) ;;
        esac
    done
}

fix(){
    sudo -v

    # Delete config_backup directory
    pos1=$(getcursor)
    loading_wheel "${Y}●${NC} Delete config_backup folder" &
    loading_pid=$!
    if [ ! -d "$HOME/$backup_folder" ]; then
        echo -e "${R}Nothing to delete: config_backup folder does exist${NC}\n"
    else
        rm -rf ~/config_backup
        tput cup $((pos1 - 1)) 0
        tput ed
        echo -e "${G}●${NC} Delete config_backup folder ${G}Done!${NC}\n"
    fi
    kill $loading_pid

[... 59 lines cut ...]
    timeout: 90.0
    verbose: True'

    if grep -rqF "$simple_klipper_macro_pattern" "$HOME/printer_data/config/" >/dev/null 2>&1; then
        echo "Klipper macro pattern found in the following files:"
        grep -rlF "$simple_klipper_macro_pattern" "$HOME/printer_data/config/"
        echo -e "\nPlease check yourself whether the structure is correct. The structure is expected as follows:\n"
        echo -e "$klipper_macro_pattern"
    fi

    # Reload systemd manager configuration
    pos1=$(getcursor)
    loading_wheel "${Y}●${NC} Reload systemd manager configuration" &
    loading_pid=$!
    sudo systemctl daemon-reload
    tput cup $((pos1 - 1)) 0
    tput ed
    kill $loading_pid
    echo -e "${G}●${NC} Reload systemd manager configuration ${G}Done!${NC}\n"

    exit
}

fix
```

shisva, 2024-08-26:
Hi there, I have the same issue with backup on boot :

`mikael@klipper:~/klipper-backup $ sudo systemctl status klipper-backup-on-boot.service
● klipper-backup-on-boot.service - Klipper Backup On-boot Service
     Loaded: loaded (/etc/systemd/system/klipper-backup-on-boot.service; enabled; vendor preset: enabled)
     Active: failed (Result: exit-code) since Mon 2024-08-26 20:41:38 CEST; 7min ago
   Main PID: 410 (code=exited, status=127)
        CPU: 39ms

Aug 26 20:41:38 klipper systemd[1]: Starting Klipper Backup On-boot Service...
Aug 26 20:41:38 klipper env[410]: bash: /klipper-backup/script.sh: No such file or directory
Aug 26 20:41:38 klipper systemd[1]: klipper-backup-on-boot.service: Main process exited, code=exited, status=127/n/a
Aug 26 20:41:38 klipper systemd[1]: klipper-backup-on-boot.service: Failed with result 'exit-code'.
Aug 26 20:41:38 klipper systemd[1]: Failed to start Klipper Backup On-boot Service.
`

tried the --fix, tried reinstall, tried your new fix just bellow and unfortunatly nothing

BTW commit on change seems working.

crowsnest
v4.1.9-1-gd75a3aeb

klipper
 v0.12.0-230 > v0.12.0-286

klipper-backup
v1.3.1-3-g46321046

mainsail
v2.12.0

mainsail-config
v1.2.1-0-ge57810d5

mobileraker
v0.4.0-77-gff0d7d70

moonraker
v0.9.2-6-gf735c041

sonar
v0.1.3-1-geacc498a

Tylerjet, 2024-08-26:
Can you screenshot or copy paste the view of the service file it should be in '/etc/systemd/system'

The on boot service should be '$HOME/klipper-backup/script.sh' but based on the log you sent the service does not see the path that $HOME should have assigned to it. 

This makes me think the user section of the service might not have been set properly.

shisva, 2024-08-27:
Hello !

here is the content of the klipper-backup-on-boot.service : 

```
[Unit]
Description=Klipper Backup On-boot Service
After=
Wants=

[Service]
User=
Type=oneshot
ExecStart=/usr/bin/env bash  -c "/usr/bin/env bash $HOME/klipper-backup/script.sh -c \"New Backup on boot - $(date +\"%%x - %%X\")\""

[Install]
WantedBy=default.target
```

Tylerjet, 2024-08-27:
Ahh so it looks like @Staubgeborener may have just forgoten a few pieces for the fix script above. The install.sh script fills in those blank sections with your username and the correct wants after sections. His adjustments currently just copy over the file but don't remake those adjustments.

Could you try the following:
```
source "$HOME/klipper-backup/utils/utils.func"

fix(){
    sudo -v

    # Delete config_backup directory
    pos1=$(getcursor)
    loading_wheel "${Y}●${NC} Delete config_backup folder" &
    loading_pid=$!
    if [ ! -d "$HOME/config_backup" ]; then
        echo -e "${R}Nothing to delete: config_backup folder does exist${NC}\n"
    else
        rm -rf ~/config_backup
        tput cup $((pos1 - 1)) 0
        tput ed
        echo -e "${G}●${NC} Delete config_backup folder ${G}Done!${NC}\n"
    fi
    kill $loading_pid

    # Check and fix klipper-backup-on-boot.service
    if [ -f /etc/systemd/system/klipper-backup-on-boot.service ]; then
        if [[ "$(md5sum $HOME/klipper-backup/install-files/klipper-backup-on-boot.service | awk '{print $1}')" != "$(md5sum /etc/systemd/system/klipper-backup-on-boot.service | awk '{print $1}')" ]]; then
            pos1=$(getcursor)
            if ask_yn "klipper-backup-on-boot.service is corrupted, should this be fixed? ${R}Attention: the klipper-backup-on-boot.service will be overwritten with the klipper-backup default settings, manual changes to this file will be overwritten!${NC}"; then
                loading_wheel "${Y}●${NC} Fix \"backup on boot\" service" &
                loading_pid=$!
                sudo cp ~/klipper-backup/install-files/klipper-backup-on-boot.service /etc/systemd/system/klipper-backup-on-boot.service
                sleep .5
                sudo sed -i "s/^After=.*/After=$(wantsafter)/" "/etc/systemd/system/klipper-backup-filewatch.service"
                sudo sed -i "s/^Wants=.*/Wants=$(wantsafter)/" "/etc/systemd/system/klipper-backup-filewatch.service"
                sudo sed -i "s/^User=.*/User=${SUDO_USER:-$USER}/" "/etc/systemd/system/klipper-backup-filewatch.service"
                sudo systemctl daemon-reload 2>/dev/null
                sudo systemctl enable klipper-backup-filewatch.service 2>/dev/null
                sudo systemctl start klipper-backup-filewatch.service 2>/dev/null
                kill $loading_pid
                tput cup $((pos1 - 1)) 0
                tput ed
                echo -e "${G}●${NC} Fix \"backup on boot\" service ${G}Done!${NC}\n"
            fi
        else
            echo -e "${G}●${NC} Check \"backup on boot\" service ${G}Done!${NC}\n"
        fi
    fi

    # Check and fix "automatic backup every 4 hours" cronjob
    # Check if Klipper-Backup is present in cron list
    if crontab -l | grep -q '/klipper-backup/script.sh'; then
        # If yes, check if the command is correct
        if ! crontab -l | grep -qF '0 */4 * * * $HOME/klipper-backup/script.sh -c "Cron backup - $(date +"%x - %X")"'; then
        echo -e "The \"automatic backup every 4 hours\" cronjob is faulty. It is:\n${Y}$(crontab -l | grep '/klipper-backup/script.sh')${NC}\nKlipper-Backup uses the following variant:\n${Y}0 */4 * * * \$HOME/klipper-backup/script.sh -c \"Cron backup - \$(date +\"%x - %X\")\"${NC}\n${NC}"
            if ask_yn "Should it be corrected? ${R}Attention: This function is experimental depending on the cron list. It is always recommended to edit the list manually${NC}"; then
                loading_wheel "${Y}●${NC} Fix \"automatic backup every 4 hours\" cronjob" &
                loading_pid=$!
                (crontab -l | grep -v '/klipper-backup/script.sh'; echo '0 */4 * * * $HOME/klipper-backup/script.sh -c "Cron backup - $(date +"%x - %X")"') | crontab -
                sleep .5
                kill $loading_pid
                tput cup $((pos1 - 1)) 0
                tput ed
                echo -e "${G}●${NC} Fix \"automatic backup every 4 hours\" cronjob ${G}Done!${NC}\n"
            fi
        else
            echo -e "${G}●${NC} Check \"automatic backup every 4 hours\" cronjob ${G}Done!${NC}\n"
        fi
    else
        echo -e "${G}●${NC} There is no \"automatic backup every 4 hours\" cronjob ${G}Done!${NC}\n"
    fi

    # Check Klipper macro
    simple_klipper_macro_pattern='/klipper-backup/script.sh'
    klipper_macro_pattern='[gcode_macro update_git]
    gcode:
        {% set message = params.MESSAGE|default() %}
        {% if message %}
            RUN_SHELL_COMMAND CMD=update_git_script_message PARAMS="'{params.MESSAGE}'"
        {% else %}
            RUN_SHELL_COMMAND CMD=update_git_script
        {% endif %}

    [gcode_shell_command update_git_script]
    command: bash -c "bash $HOME/klipper-backup/script.sh"
    timeout: 90.0
    verbose: True

    [gcode_shell_command update_git_script_message]
    command: bash -c "bash $HOME/klipper-backup/script.sh -c $0"
    timeout: 90.0
    verbose: True'

    if grep -rqF "$simple_klipper_macro_pattern" "$HOME/printer_data/config/" >/dev/null 2>&1; then
        echo "Klipper macro pattern found in the following files:"
        grep -rlF "$simple_klipper_macro_pattern" "$HOME/printer_data/config/"
        echo -e "\nPlease check yourself whether the structure is correct. The structure is expected as follows:\n"
        echo -e "$klipper_macro_pattern"
    fi

    # Reload systemd manager configuration
    pos1=$(getcursor)
    loading_wheel "${Y}●${NC} Reload systemd manager configuration" &
    loading_pid=$!
    sudo systemctl daemon-reload
    tput cup $((pos1 - 1)) 0
    tput ed
    kill $loading_pid
    echo -e "${G}●${NC} Reload systemd manager configuration ${G}Done!${NC}\n"

    exit
}

fix
```

shisva, 2024-08-28:
Hi  @Tylerjet , I had to remove a "fi" on ligne 82

then when I execute it the rotating bar is spinning without ending.

However, the file is edited :
```[Unit]
Description=Klipper Backup On-boot Service
After=network-online.target
Wants=network-online.target

[Service]
User=mikael
Type=oneshot
ExecStart=/usr/bin/env bash  -c "/usr/bin/env bash $HOME/klipper-backup/script.sh -c \"New Backup on boot - $(date +\"%%x - %%X\")\""

[Install]
WantedBy=default.target
```

and when i run ```sudo systemctl status klipper-backup-on-boot.service ```I have this, btw the status is ```activating (start)``` when other macro have their satus ```active```

```mikael@klipper:~ $ sudo systemctl status klipper-backup-on-boot.service
[sudo] password for mikael:
● klipper-backup-on-boot.service - Klipper Backup On-boot Service
     Loaded: loaded (/etc/systemd/system/klipper-backup-on-boot.service; enabled; vendor preset: enabled)
     Active: activating (start) since Wed 2024-08-28 20:22:38 CEST; 1min 22s ago
   Main PID: 624 (bash)
      Tasks: 5 (limit: 8544)
        CPU: 1min 43.516s
     CGroup: /system.slice/klipper-backup-on-boot.service
             ├─624 bash /home/mikael/klipper-backup/script.sh -c New Backup on boot - 28/08/24 - 20:22:38
             ├─820 bash /home/mikael/klipper-backup/script.sh -c New Backup on boot - 28/08/24 - 20:22:38
             ├─821 cat /dev/urandom
             ├─822 tr -dc a-z0-9
             └─823 fold -w 7

Aug 28 20:22:38 klipper systemd[1]: Starting Klipper Backup On-boot Service...
Aug 28 20:22:38 klipper env[644]: [42B blob data]
Aug 28 20:22:38 klipper env[624]: [49B blob data]
Aug 28 20:22:41 klipper env[624]: Klipper-backup is up to date
Aug 28 20:22:41 klipper env[813]: Initialized empty Git repository in /home/mikael/config_backup/.git/
```

Tylerjet, 2024-08-29:
Oh woops yeah forgot to remove the extra fi, not sure why it didn't kill the loading wheel though (but not a big deal at the moment).

AFAIK "activating" should be the proper state as it should then become inactive/active after its run on bootup and it seems to have ran without errors as far as i can see.

Jaizu, 2024-08-29:
I was kinda busy lately and didn't play much with the 3d printer aside to print some quick stuff that I needed. I upgraded everything on the system a few days ago and klipper-backup seems to have fixed itself alone during that update, at least I think so.
[image]

Staubgeborener, 2024-08-29:
Ok let's close this issue. The `--fix` option will be adjusted in future anyway.