Staubgeborener/Klipper-Backup #118: Klipper back up error

coros25, 2024-08-12:
I am new to using the klipper backup command and when I run the macro, I see the following errors.
Here is the terminal log when running the macro:

10:58 AM
Command {update_git_script} finished
10:58 AM
hint: Updates were rejected because the tip of your current branch is behind
hint: its remote counterpart. Integrate the remote changes (e.g.
hint: 'git pull ...') before pushing again.
hint: See the 'Note about fast-forwards' in 'git push --help' for details.
10:58 AM
error: failed to push some refs to 'https://github.com/coros25/Voron0.2.git'
10:58 AM
To https://github.com/coros25/Voron0.2.git
! [rejected]        main -> main (non-fast-forward)
10:58 AM
From https://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@github.com/coros25/Voron0.2.git
10:58 AM
On branch main
nothing to commit, working tree clean
10:58 AM
Skipping symbolic link: printer_data/config/timelapse.cfg
10:58 AM
Skipping symbolic link: printer_data/config/mainsail.cfg
10:58 AM
Skipping symbolic link: printer_data/config/KAMP
10:58 AM
fatal: refusing to merge unrelated histories
10:58 AM
From https://github.com/coros25/Voron0.2
* branch            main       -> FETCH_HEAD
10:58 AM
hint: Pulling without specifying how to reconcile divergent branches is
hint: discouraged. You can squelch this message by running one of the following
hint: commands sometime before your next pull:
hint:
hint:   git config pull.rebase false  # merge (the default strategy)
hint:   git config pull.rebase true   # rebase
hint:   git config pull.ff only       # fast-forward only
hint:
hint: You can replace "git config" with "git config --global" to set a default
hint: preference for all repositories. You can also pass --rebase, --no-rebase,
hint: or --ff-only on the command line to override the configured default per
hint: invocation.
10:58 AM
Klipper-backup is up to date
10:58 AM
[1;93m●[0m Checking for installed dependencies /
[K[1;92m●[0m Checking for installed dependencies [1;92mDone![0m
10:58 AM
Running Command {update_git_script}...:
10:58 AM
update_git

Staubgeborener, 2024-08-12:
I guess eother one of these two things:

1. you didn't follow the [documentation](https://klipperbackup.xyz/)
2. or you try to push the files to a repository which already exists 

The error message is about having a repository at GitHub which has a different commit history as the local one. Mostly this happens while trying to push a commit to a repository which already exists with files.

So I would suggest the following: try `./script.sh --fix` and see if you can run the script afterwards without the `--fix` parameter.

If this doesn't work, try to push your backup to another repository. So create a new one on GitHub with a new token, rerun the installation script with the new information and push your backup to the new repository. If this works, delete the old repository because you will not need this anymore.

coros25, 2024-08-16:
Yep you were right! Works like a charm just creating a new repository.  Thank you so much!

closed 2024-08-16 by coros25