Staubgeborener/Klipper-Backup #120: Script rm -rf'ed my entire home dir...

CactusBoy3, 2024-08-18:
### Code of Conduct

- [X] I promise that i have used markdown syntax to make the `code` more readable which helps to fix the problem and that I have tested the `--fix` command mentioned above without success.

### What happened

[image]
my whole home dir is gone... this script did the exact opposite of what it was supposed to 

### What did you expect to happen

i expected it to not delete my files

### How to reproduce

idk

Staubgeborener, 2024-08-18:
Well, this has **definitely** nothing to do with this backup script. But I'm curious, please show me the output/content of your `.bash_history` file so we can see what you actually did in your shell.

Staubgeborener, 2024-08-18:
I'm curious, please show me the output/content of your `.bash_history` file so we can see what you actually did in your shell.

The only line which is running a `rm -rf` is [this one](https://github.com/Staubgeborener/Klipper-Backup/blob/46321046222c1a707873a24eb18b9ea5377cf815/script.sh#L132), but this only deleted the content inside the `$backup_path` variable which is [$HOME/config_backup](https://github.com/Staubgeborener/Klipper-Backup/blob/46321046222c1a707873a24eb18b9ea5377cf815/script.sh#L22), so my first thought: you changed this line. Also it's weird that you show the content of your `/home/user/ethan52` which has a `.git` folder - is this one the `.git` folder of Klipper-Backup or why do you have a `.git` folder in your home directory?

I would simply love to know how you tried to install Klipper-Backup. I'm just trying to understand it with the information you provided: there is a `.git` folder and a `Readme.md` file in your home directory. Maybe those belongs to Klipper-Backup, maybe not. Let's assume yes. Klipper-Backup deletes the `config_backup` after running the script except `.git` and `Readme.md` - that would makes sense, because those are the only left files from the script. But those files belongs in a separate `config_backup` folder and not in you home directory. Klipper-Backup does this automatically. So it looks like you just drag and drop this code onto your sd card, didn't read the docs at all and also edit [line 22](https://github.com/Staubgeborener/Klipper-Backup/blob/46321046222c1a707873a24eb18b9ea5377cf815/script.sh#L22). Is this true? Again I would love to see this `.bash_history` file.

CactusBoy3, 2024-08-18:
backup_folder="config_backup"
backup_path="$HOME/$backup_folder"

was changed to 

backup_folder="ender5_data"
backup_path="$HOME/$ender5_data"

my thinking was that it would change the backup directory. evidently it did not. :/ 

contents of .bash_history

```
##running command
ls
~/klipper-backup/script.sh
~/klipper-backup/install.sh
ls
##realising what i had done
cd ~/
ls
nano readme.md
ls
cd media
ls
cd plex
sudo shutdown now
```

I had downloaded the script using the curl command and then executed the script with the other command. I gave it all the requested info and it uploaded files to GitHub
[image]
It did not upload any config files tho. I looked around the wiki but somehow missed the "Paths" section. I leave it for a while and then come back to it. I edited the script.sh as I thought that was the correct way to change the path. I run the script then boom. Its my fault really :/.  The .git folder and Readme.md are from the script.

Staubgeborener, 2024-08-18:
> backup_folder="config_backup"
> backup_path="$HOME/$backup_folder"
> 
> was changed to
> 
> backup_folder="ender5_data"
> backup_path="$HOME/$ender5_data"

And this is the reason why your files are gone.

The backup path was `$HOME/$ender5_data` (which is `/home/user/ender5_data`). After running the script, it nukes the content out of the `$backup_folder` which was sadly your home directory. Sorry to say that, but the files are gone because of this edit.

CactusBoy3, 2024-08-18:
:/ prob should have run it in a vm or something first rather on the pi with all my files on. i really should make backups of important files.  https://www.youtube.com/watch?v=tu_wTvstWcw

closed 2024-08-18 by CactusBoy3

Tylerjet, 2024-08-19:
> backup_folder="config_backup"
> backup_path="$HOME/$backup_folder"
> 
> was changed to
> 
> backup_folder="ender5_data"
> backup_path="$HOME/$ender5_data"

Also just to add here, the reason why your home directory was deleted was due to your change of the second line. if you would have left it as `$HOME/$backup_folder` then it would have set the path as `$HOME/ender5_data`. 

But you changed it to be `$ender5_data`, which bash sees as a variable, but that variables data does not exist and thus is empty which made your backup_path be seen by bash as just `$HOME/`. Since $ender5_data is a empty variable.