Staubgeborener/Klipper-Backup #122: Klipper-Backup using 100% CPU, throttling raspberry pi temps over 80C

voron69-bit, 2024-09-02:
### Code of Conduct

- [X] I promise that i have used markdown syntax to make the `code` more readable which helps to fix the problem and that I have tested the `--fix` command mentioned above without success.

### What happened

Root Cause Analysis: High CPU Usage in Klipper Backup Script
Issue Description:

A significant spike in CPU usage was observed on a Raspberry Pi 4B when executing the Klipper backup script (/home/pi/klipper-backup/script.sh). at startup The issue was traced to three specific processes: cat, tr, and fold, which were consuming an unusually high amount of CPU resources.

Technical Findings:

Process Identification:

The processes consuming high CPU were identified as:
cat /dev/urandom
tr -dc 'a-z0-9'
fold -w 7
These processes were found to be continuously running, leading to sustained high CPU usage.
Parent Process:

The parent process of these commands was traced to a Bash script located at /home/pi/klipper-backup/script.sh.
The specific command sequence was used to generate a random string for inclusion in an email address configuration during the script’s execution.
Impact:

The continuous reading and processing of data from /dev/urandom caused a high CPU load, leading to system performance degradation and extreme Pi temps ( 85C ) at printer idle.
The issue was isolated to the random string generation process within the backup script, which was observed to be resource-intensive on the Raspberry Pi 4B.
Behavior Across Systems:

This behavior was noted to be inconsistent across different systems, with the issue specifically manifesting on a Raspberry Pi 4B while not being observed on another system running a similar setup.
Conclusion:

The high CPU usage was directly linked to the random string generation process within the Klipper backup script, specifically involving the use of cat, tr, and fold commands. The continuous data processing from /dev/urandom was identified as the primary cause of the excessive CPU load.

### What did you expect to happen

That this wouldn't happen. Running fine on my other printer.

### How to reproduce

Reboot or rerun the script.

### Additional information

The fix script did not fix the issue. Once I kill cat PID after rebooting, the remainder of the tasks end along with it, and the printer behaves normally.

Staubgeborener, 2024-09-02:
What OS are you running on the raspberry pi 4? Is it possible that you have updated it recently and the problem has arisen since then? As far as I know, the raspberry pi 4 has been notorious for quickly producing pointless CPU spikes and temperature rises (one of the reasons why I sold mine again after just a few months).

Otherwise I can look into it, but for testing purposes I only have a raspberry pi 3b, BTT pi and some intel/AMD devices. No guarantee, as this is a very hardware-specific problem.

Staubgeborener, 2024-09-02:
What OS are you running on the raspberry pi 4? Is it possible that you have updated it recently and the problem has arisen since then? As far as I know, the raspberry pi 4 has been notorious for quickly producing pointless CPU spikes and temperature rises (one of the reasons why I sold mine again after just a few months). What OS are you running? 

Otherwise I can look into it, but for testing purposes I only have a raspberry pi 3b, BTT pi and some intel/AMD devices. No guarantee, as this is a very hardware-specific problem. In the meanwhile you could try to replace [L111](https://github.com/Staubgeborener/Klipper-Backup/blob/9d049cc1aa0463b8bb1937613f3431ffbac19fa5/script.sh#L111) with `unique_id=$(echo $RANDOM$RANDOM | tr -dc 'a-z0-9' | fold -w 7 | head -n 1)` so we get rid of `/dev/urandom` in order to exclude this snippet as a theoretical problem.

voron69-bit, 2024-09-02:
> What OS are you running on the raspberry pi 4? Is it possible that you have updated it recently and the problem has arisen since then? As far as I know, the raspberry pi 4 has been notorious for quickly producing pointless CPU spikes and temperature rises (one of the reasons why I sold mine again after just a few months).
> 
> Otherwise I can look into it, but for testing purposes I only have a raspberry pi 3b, BTT pi and some intel/AMD devices. No guarantee, as this is a very hardware-specific problem. In the meanwhile you could try to replace [L111](https://github.com/Staubgeborener/Klipper-Backup/blob/9d049cc1aa0463b8bb1937613f3431ffbac19fa5/script.sh#L111) with `unique_id=$(date +%s%N | md5sum | head -c 7)` so we get rid of `/dev/urandom` in order to exclude this snippet as a theoretical problem.

I did try this already, and it appears to have fixed the runaway process with cat, but the actual backup is not working.
I'm now getting this when I manually run the script:

`pi@voron:~ $ bash ~/klipper-backup/script.sh --commit_message "Test Backup"
● Checking for installed dependencies Done!

Klipper-backup is up to date

fatal: not a git repository (or any of the parent directories): .git
fatal: not in a git directory
fatal: not in a git directory
fatal: not a git repository (or any of the parent directories): .git
fatal: not a git repository (or any of the parent directories): .git
fatal: not a git repository (or any of the parent directories): .git
fatal: not a git repository (or any of the parent directories): .git
fatal: not a git repository (or any of the parent directories): .git
fatal: not a git repository (or any of the parent directories): .git
fatal: No remote configured to list refs from.
fatal: not a git repository (or any of the parent directories): .git
fatal: not a git repository (or any of the parent directories): .git
`

voron69-bit, 2024-09-02:
I recently updated all raspberry OS, Klipper, Mainsail, Crowsnest, etc to latest.

Staubgeborener, 2024-09-03:
> I did try this already, and it appears to have fixed the runaway process with cat

Ok so the initial problem is solved.

About your other one: please show the output of

1. `ls -la ~/config_backup`
2. `cat ~/config_backup/.git/config`
3. `~/klipper-backup/.env` (hide the token)

And post a link of your github repository where you want to push your backup.

voron69-bit, 2024-09-03:
I ended up reinstalling. After fighting with klipper-backup in moonraker saying "INVALID" for nearly two hours, I THOUGHT I fixed it, but turns out, I re-generated the issue! So when I completely removed Klipper-backup, and reinstalled, it worked perfect. WITHOUT autobackup and Moonraker Updating.

Moonraker said INVALID, so I did the following:
Removed all of [update_manager klipper-backup] from moonraker.conf
sudo rm -rf /home/pi/printer_data/config/moonraker.json
sudo systemctl restart moonraker
sudo rm -rf ~/klipper-backup
git clone https://github.com/Staubgeborener/klipper-backup.git ~/klipper-backup

Readded:

[update_manager klipper-backup]
type: git_repo
path: ~/klipper-backup
origin: https://github.com/Staubgeborener/klipper-backup.git
managed_services: moonraker
primary_branch: main

to moonraker.conf
Rebooted, CAT etc running amuck.

Staubgeborener, 2024-09-03:
Of course moonraker says invalid (or give you the dirty flag) because you edited a line (L111) manually thus this code doesn't match with this repository anymore. Tbh I can only help you would answer my question and give Ben time the asked output, right now you just "do something" and ignore my comments which is not very productive.

Staubgeborener, 2024-09-03:
Of course moonraker says invalid (or give you the dirty flag) because you edited a line (L111) manually thus this code doesn't match with this repository anymore. Tbh I can only help if you would answer my questions and give the asked output, right now you just "do something" and ignore my comments which is not very productive.

Staubgeborener, 2024-09-03:
Of course moonraker says invalid (or give you the dirty flag) because you edited a line (L111) manually thus this code doesn't match with this repository anymore. You can safely ignore this error as I will add the new unique id generation in the next days. After that you can update Klipper-Backup (also with a hard-reset/soft-reset via moonraker). But until then you would get rid of the initial issue.
Tbh I can only help if you would answer my questions and give the asked output, right now you just "do something" and ignore my comments which is not very productive.

voron69-bit, 2024-09-03:
> Of course moonraker says invalid (or give you the dirty flag) because you edited a line (L111) manually thus this code doesn't match with this repository anymore. You can safely ignore this error as [...]

Incorrect, I deleted klipper-backup entirely, and re-cloned it. So L111 was back to using cat. After some sleep, I deleted the klipper-backup again, re-cloned it, re-created the .env file, rebooted, and those three pids aren't running the pi into the ground anymore, and moonraker isn't complaining about invalid versions. Unfortunately I am unable to give you further details for RCA, just know that there is an issue somewhere with those three jobs taking over the box if certain issues arise, which is not good.

closed 2024-09-03 by voron69-bit